dotnet/iot #72: System.Devices.Gpio needs xml doc comments

joshfree, 2018-12-04:


shaggygi, 2019-01-23:
@joshfree @joperezr If time allows, I'll try to review and add summaries for PWM code next.  Is there something else outstanding for this I should focus on to consider this complete?

joperezr, 2019-01-23:
Thanks so much for all the hard work you've been doing for this issue @shaggygi, I believe you are right and PWM is the only area we are missing now. For the device-bindings we are treating xml-comments as a nice-to-have right now but we aren't enforcing them, so this issue is only set to cover the System.Device.Gpio library.

shaggygi, 2019-01-23:
No prob.  It's been a fun exercise and helped learn more about the inner-workings of the API along the way.  Hope do help lots more in future, as well. :smile:

shaggygi, 2019-02-05:
We should be able to close this one now.

joperezr, 2019-02-05:
Agreed, thanks for all the hard work @shaggygi

closed 2019-02-05 by joperezr