dotnet/iot #78: Virtual devices & signalr

grahamehorner, 2018-12-05:
Please consider creating a set of virtual devices that simulate the physically state/operations of the hardware, and can be used without the need of the developer to have access to actual hardware.

E.g.. the device driver for a virtual SPI interface could make use of Signalr to send (almost realtime) state updates and/or operations to a web, windows, mobile ux.
using a similar concept to device twins.

IMHO this would help developers and improve testibility and reduce the developer inner loop when creating driver/controllers.

joperezr, 2018-12-05:
I agree that having this would be great. TBH I'm not super familiar with SignalR so I don't know how costing would look like to have this, but it would definitely help inner-loop and validation since you would be able to have tests without relying on the hardware.

grahamehorner, 2018-12-05:
@joperezr I'd be happy to do a proof of concept for a virtual GPIO based around signalr/websockets; to see if this is possible and/or is a good fit for the developer inner loop/testing without the need for hardware.

joperezr, 2018-12-05:
That would be great! 😄

grahamehorner, 2018-12-06:
@joperezr I've start looking into this and I'm building up a set of IObservable and IObserver classes that will represent the physical components within a circuit and the voltage/current etc.

FYI, I'm looking to have a basic ADC virtual circuit working by mid/end of next week; I'm dug out an olde copy of a book I had used many year ago Electronic Circuit & System Simulation Methods (SRE) by Lawrence Pillage

shaggygi, 2019-03-18:
@krwq @joperezr  I've started playing with this repo ([Project TIOT](https://github.com/shaggygi/Tiot)).  It's basically a dumping ground for me trying to learn about related topics.  However, it might help with this issue.  It is still "very" early, but looks promising.  I was already able to perform many of the GPIO-related features and was successful communicating with I2C (but is far as I got for now).  There is a [basic example using with a WPF app](https://github.com/shaggygi/Tiot/tree/master/samples/Tiot.Grpc/Tiot.Grpc.GpioWithWpf), as well.

If it gains traction/support, it might be a good idea to move over to the [tools section](https://github.com/dotnet/iot/tree/master/tools).

Again, this is nothing more than a playground for myself learning topics and repo could be removed at some point.  Just thought you might be interested.

Much more to do as time allows. 🎉

krwq, 2019-03-18:
@shaggygi perhaps you could use Firmata to do stuff remotely: https://github.com/dotnet/iot/issues/262 - I bet same stuff will work for Pi and Arduino.

For many scenarios TCP connection will do but in some cases you might have to send a program which executes remotely (I don't know if firmata has capabilities like that but might be worth taking a look since it seems it has similar goals in mind)

shaggygi, 2019-03-18:
@krwq Thanks for the link.  I had not come across that one yet.  It seems this is the same concept I'm trying to do with TIOT, except Firmata appears to use serial/USB (which is still a good idea) instead of gRPC & Ethernet.  I'll try to read over the content/topic.  Thx

krwq, 2019-03-18:
@shaggygi if we decide to implement it, might be worth to just depend on Stream and then you can either pass a serial stream or tcp/other stream depending on what you need (perhaps we can have an overload which uses serial by default as well but stream should be main API)

krwq, 2021-01-07:
[Triage] How should do APIs look like for this?

Ellerbach, 2021-07-01:
[Triage] We now have multiple way of doing this:
-Arduino with Firmata support
- FT4222 thru USB
- For tests, we have multiple example of mocking up SPI, I2C devices.

Question is: is this issue still valid?

closed 2021-07-01 by Ellerbach