dotnet/iot #79: I2C data-address

deinok, 2018-12-05:
Im using https://linux.die.net/man/8/i2cset in order to access to leds. But I would like to change to use `UnixI2cDevice`  

My question is how do I set the `data-address` of write?

joperezr, 2018-12-05:
If by this you mean the I2c Address, then this is set in the `I2cConnectionSettings.DeviceAddress`. The way it works is that you would generate a I2cConnectionSettings object, and then pass that in to the constructor of the UnixI2cDevice, for example:
```c#
var settings = new I2cConnectionSettings(busId: 0, deviceAddress: 0x77);
var myI2cDevyce = new UnixI2cDevice(settings);
```

deinok, 2018-12-05:
@joperezr In my case I use for now the following `i2cset -y 1 0x69 0x00 0xFF && i2cset -y 1 0x69 0x01 0x00 && i2cset -y 1 0x69 0x02 0x00` for set a led to red.
As you can see, 1 is busId, 0x69 is for deviceAddress, 0x00, 0x01, 0x02 is for registers or data-address and at the end we have the RGB value.

I'm not sure, but I think that writting a byte array with the register and value works well. Could we add a method to simplificate this?

deinok, 2018-12-05:
@joperezr Yes, thanks. In my case I will add an extension method for my specific device.

closed 2018-12-05 by deinok

joperezr, 2018-12-05:
cool. I plan on adding some Spi and I2c device bindings as an example of the API consumption soon 😄

deinok, 2018-12-06:
Cool. If you need it, I could give bindings for https://www.access-is.com/products/val100