dotnet/iot #84: Create a standard binding sample template

shaggygi, 2018-12-07:
It would be nice to have a standard outline/sections for binding samples.  This could be a markdown stored in repo for new READMEs to be copied into new project is added.  I'm assuming the Mcp3008 README could be uses as a beginner (assuming we would remove specific content (links, Fritzing file, etc.) before moving the standard template markdown.

Is there anything needed to be added/removed in Mcp3008 README today?  Any sections that might be helpful for other types of bindings?

joperezr, 2019-02-11:
Do we really need to ship this binding as a NuGet Package? Is there a NuGet convention for project templates? IMHO I would consider this issue done, given that the main consumers of the template will be people contributing to our repo, so installing it from the local path seems like a good enough solution to me. What do you think @shaggygi ?

shaggygi, 2019-02-11:
No, a package isn't really needed and users can install as is.  Markdown should explain enough to get them going.  I'll go ahead and close.

closed 2019-02-11 by shaggygi

joperezr, 2019-02-11:
BTW, this was used a lot during the hackathon and everyone that did pointed out how useful it was to get started with it, so just extending the kudos to you who actually did all the work.

shaggygi, 2019-02-11:
Thx.  You pointed out a few goodies to make it better, as well :+1: