dotnet/iot #88: Create a Microchip MCP23XXX binding

shaggygi, 2018-12-09:
Create a binding API for the Microchip MCP23XXX family.

//cc @joperezr @joshfree Please assign this to me.  I have a prototype, but trying to work out the kinks and such.  This has been a good exercise as it has bought up a few issues/questions.  I will post more details and action items as progress is made.  Thx

shaggygi, 2018-12-11:
See initial PR https://github.com/dotnet/iot/pull/91

shaggygi, 2018-12-16:
@joperezr @joshfree Would it be possible to go ahead and check-in https://github.com/dotnet/iot/pull/91?  This issue can continue to stay open, though.

This currently works for SPI and has some goodness for a starting layout on binding docs.

I've been thinking about binding fundamentals and understand this will change in future.  For example, similar to the Mcp2008 binding and implementing https://github.com/dotnet/iot/issues/90, I could see where I/O expanders could implement a similar format.  I'm not sure if this would be a [type of bus provider](https://docs.microsoft.com/en-us/windows/iot-core/develop-your-app/busproviders), but warrants more thoughts before getting too deep in developing more bindings.

Thx

joperezr, 2018-12-17:
@shaggygi I left some feedback on the PR and it looks like you haven't had the chance yet to address/respond to it. Would you mind taking a look?

closed 2019-02-08 by shaggygi