dotnet/iot #93: Productize the device bindings and samples from corefxlab

joshfree, 2018-12-12:
@edgardozoppi created a bunch of samples this summer under https://github.com/dotnet/corefxlab/tree/master/samples/System.Devices.Gpio.Samples. 

The device bindings should be moved to the Iot.Device namespace under https://github.com/dotnet/iot/tree/master/src/devices

The Application code samples using the device bindings should be moved under https://github.com/dotnet/iot/tree/master/samples.

[Fritzing Diagrams](http://fritzing.org/home/) should be generated liberally.

krwq, 2019-02-22:
https://github.com/dotnet/corefxlab/tree/master/samples/System.Devices.Gpio.Samples doesn't have anything left there. Is there any more work here or should we close?

buyaa-n, 2019-02-22:
Seems all done except 
> Fritzing Diagrams should be generated liberally.

Not sure if it is separate task to add diagrams for all samples which is missing a diagram

krwq, 2019-02-22:
I think we should have separate issue for that - seems like there is nothing left in the corefxlab. Let's reopen if someone reports we missed something.

Btw. I don't think all devices need the diagram (I2C/SPI and the hat devices are pretty much the same for all so for those I don't think it is needed)

closed 2019-02-22 by krwq