dotnet/iot #107: [Proposal] Add System.Device.Gpio.Toggle()

josesimoes, 2018-12-17:
Toggling a Gpio pin is a very, very common operation. To accomplish that, with what's available today, one has to either:
1. Keep a variable to hold the pin state and then update the pin _and_ the variable.
2. Perform a read pin, toggle state and perform a write pin.

Most of modern MCUs provide a low level way of doing this with one or maybe a couple of instructions. Even if it requires implementing 2. in native code it would still be far more efficient that using 1. or 2. in managed code.

Considering the above, I'm proposing that a `System.Device.Gpio.Toggle(int pinNumber)` method should exist.

PS: We already have this on [nanoFramework](https://github.com/nanoframework/) Windows.Devices.Gpio API, see [here](http://docs.nanoframework.net/api/Windows.Devices.Gpio.GpioPin.html#Windows_Devices_Gpio_GpioPin_Toggle).

grahamehorner, 2018-12-17:
a FlipFlop operation is also very common aka alternate between states (on/off) for x cycles.

shaggygi, 2018-12-17:
@grahamehorner FlipFlop seems to more of a type of circuit (https://en.wikipedia.org/wiki/Flip-flop_(electronics)).  I could eventually see a bindings for a [7474-type component](http://www.ti.com/lit/ds/sdls119/sdls119.pdf), though.

joperezr, 2018-12-20:
The part I don't like about Toggle is that it means that we have to keep some sort of state around, which makes it dangerous for multi-threaded scenarios. If you wanted to implement this thread-safe, then you would be pretty much having to do a read and then a write inside the Toggle method, which in my mind won't add much value given that as you said you can already do this yourself by writing a small helper function.

shaggygi, 2018-12-20:
Just throwing this out as an option for conversation.  Not saying it is the best (actually causes more overhead) and probably simpler to continue using Write methods for this.

``` csharp

public PinValue Toggle(int pinNumber, PinValue currentValue)
{
    PinValue toggledValue;

    if(currentValue == PinValue.Low)
    {
        toggledValue = PinValue.High;
    }
    else
    {
        toggledValue = PinValue.Low;        
    }

    _controller.Write(pinNumber, toggledValue);
    return toggledValue;
}

// So used in example....
int pinNumber = 2;
PinValue currentValue = PinValue.Low;
_controller.OpenPin(clkPin, PinMode.Output);  // Setup clock pin.
_controller.Write(pinNumber, currentValue); // Set to initial low state.

// Do work...

// And toggle...
currentValue = Toggle(pinNumber, currentValue);  // Will toggle high.
currentValue = Toggle(pinNumber, currentValue);  // Will toggle low.
```

josesimoes, 2018-12-20:
@joperezr 

> it means that we have to keep some sort of state around

that won't be necessary, because the state is kept at the CPU level in the Gpio register.

The descriptions of "pseudo-code" I added to the issue are the usual way of dealing with this at the application. I don't have an extension for this nor anything similar.

My suggestion is for the toggle operation to be performed at the PAL level, right were the Gpio CPU registers are written. This won't require any state keeping and it's as thread safe as the existing Read or Write operations.

For CPUs that have a specific operation to toggle a Gpio pin, that would be called. For the others it would have to be mimicked with a simple 1) read Gpio register for current state 2) toggle value 3) write CPU Gpio register to update with the toggled value.

Is this more clear now? 😃

joperezr, 2018-12-20:
Right, what I meant is that if you would want to not keep state then you have to perform the two operations (read first, then write). Which CPUs have the functionality for toggling the value? AFAIK, the only one that "has" the functionality is the Windows IoT driver which if I'm not mistaken does a read and write internally anyways. All of our other existing drivers (UnixDriver, RaspberryPi driver and HummingBoard Driver) would have to do a read and write, unless we wanted to keep state.

shaggygi, 2018-12-27:
Another example...

```csharp
private void Toggle(int pinNumber, ref PinValue currentValue)
{
  if (currentValue == PinValue.Low)
  {
    currentValue = PinValue.High;
  }
  else
  {
    currentValue = PinValue.Low;
  }
  
  _controller.Write(pinNumber, currentValue);
}
```

joperezr, 2019-02-11:
I still don't think that this method would provide enough value for the potential cost (in some drivers this would require either a read/write operation, or it might also require some state to be preserved). Also, this could be easily implemented on a consumer's app in case they really need this functionality. I would like to see more feedback regarding this in order to consider it has enough value for more scenarios.

shaggygi, 2019-02-11:
I like this one better than Clock() proposal.  However, I can only recall using it in with the GpioSpiDevice proposal (#122).  I'd go ahead and close for now.

krwq, 2019-02-22:
toggle can be usually done with:
`foo ^= 0b0001000;` if we are able to do something like that directly then this should be atomic if not then I got mixed feelings if we should have it

joperezr, 2019-04-15:
I agree with @krwq. I'll go ahead and close it since there doesn't seem to be something that would add a lot of value. Feel free to re-open if there are any new cases that would make this enhancement more valuable.

closed 2019-04-15 by joperezr