dotnet/iot #111: WriteRead using I2C and MAX7311

bigdnf, 2018-12-21:
I’m trying to communicate with MAX7311 using UnixI2cDevice on Raspians. Generally I2C with other devices works but in this case I need to write some data and then immediately read response from MAX7311 device. 
When I try to do something like this:

```
var buffer = new Span<byte>();
var device = new System.Device.I2c.Drivers.UnixI2cDevice(new I2cConnectionSettings(BusId, address));
device.Write(new byte[] { 0x00 });
device.Read(buffer);

```
I get exception “Error when attempting to perform the I2c data transfer”
When I use some custom implementation of I2C from this repo

https://github.com/chkr1011/Wirehome.Core/blob/master/Wirehome.Core/Hardware/I2C/I2CBusService.cs

and use WriteRead method I can communicate with success. 

 ```
var device = new LinuxI2CBusAdapter(Logger);
 device.Enable();
 var readBuffer = new byte[2];
 device.WriteRead(address, new byte[] { 0x00 }, readBuffer);
```

It would be nice if this WriteRead could be supported – or I’m doing something wrong?

shaggygi, 2018-12-30:
I didn't look into this much, but you might need to set the buffer size to 1.  Not completely sure.  Also, make sure the BusId is correct.  I believe it should be 1.

```csharp
var device = new System.Device.I2c.Drivers.UnixI2cDevice(new I2cConnectionSettings(BusId, address));
device.Write(new byte[] { 0x00 });
byte[] readBuffer = new byte[1];
device.Read(buffer);
```

joperezr, 2019-01-09:
Because I2C protocol only uses 1 data pin, there is not a method like the TransferFullDuplex that we have in SPI. Internally what we are doing is to just do ioctl on the I2c device and send the information, so I would expect that as long as you set the right size of the read and write buffers that it should work correctly. Did you try what @shaggygi suggested?

shaggygi, 2019-01-13:
I tested this (only on Unix) and can repro this when sending Write or Read commands with an empty array or Span as shown in @bigdnf example.

@joperezr I don't understand why the ioctl wouldn't take 0 bytes, but I guess it is expecting something when doing a write/read function.  To fix this, we could put a check in `ReadWriteInterfaceTransfer` and `FileInterfaceTransfer` to just return if `writeBufferLength` or `readBufferLength` are both 0.

I don't like this as it is a little more overhead for checking on every Write/Read call and if someone used an empty array/Span... no error would be shown, the buffer would still be empty and you'd probably see another GitHub issue created stating the I2C method wasn't working.

In addition, when cleaning/updating the I2C and SPI summaries in PRs #143/#144, I added comments for buffer size when reading.  This would probably help a little.

Close issue as is?

bigdnf, 2019-01-17:
Thanks for info - I'm currently on business trip so I can't test anything. I will do it after return and close issue if I can fix it after your suggestions.

joperezr, 2019-02-12:
@bigdnf have you had the chance if @shaggygi suggestions work in your scenario?

> I tested this (only on Unix) and can repro this when sending Write or Read commands with an empty array or Span similar in @bigdnf example.

I don't see that in the example @bigdnf is using an empty array or Span, so that might be a different issue.

> To fix this, we could put a check in `ReadWriteInterfaceTransfer` and `FileInterfaceTransfer` to just return if `writeBufferLength` and `readBufferLength` are both 0.

I believe that this is an actual bug if we aren't checking the size, and if so, I would probably even throw an exception maybe if the read buffer or writebuffer is 0 as this shouldn't be allowed.

bigdnf, 2019-02-13:
@joperezr sorry for my delay with testing this - I will try to give feedback as soon as possible.

joperezr, 2019-02-13:
no worries at all, just wanted to check if that was something you still had in your plans 😄

krwq, 2019-02-22:
I think the correct fix should be to throw a descriptive exception when someone passes 0-length buffer to Read or am I missing something?

shaggygi, 2019-02-23:
@krwq The original issue was because there were no bytes allocated to perform read.  Your suggestion with having a better exception description would help in this case.  Other than that, I would label issue as stale and close.

closed 2019-02-26 by krwq