dotnet/iot #112: [Question] What is the need for OS driver files for other OS

shaggygi, 2018-12-21:
I noticed the following I2C/SPI driver files with PlatformNotSupportedException.

- UnixI2cDevice.Windows.cs
- Windows10I2cDevice.Linux.cs
- UnixSpiDevice.Windows.cs
- Windows10SpiDevice.Linux.cs

I get various ambiguous warnings/errors in Visual Studio for project.  What is the need for these and where are they used?  Just curious :thinking:

joperezr, 2019-01-07:
If you see the structure of our package, we basically have three dlls on it:
ref/netstandard2.0/
  System.Device.Gpio.dll     (<-- This assembly is the reference assembly people will compile against when using our package)
runtimes/linux/lib/netstandard2.0/
  System.Devices.Gpio.dll   (<-- This assembly will be used when running the app on linux)
runtimes/win/lib/netstandard2.0/
  System.Devices.Gpio.dll   (<-- This assembly will be used when running the app on Windows)

If we wanted to do what you point out above (making the decision at runtime of which platform to use) we would have a few problems over the current design:
 - Assemblies would become larger and larger with the support of new platforms
 - Runtime will be a bit slower as we would have to do extra checks to see which codepath to use.
 - Assemblies might end up having extra dependencies that won't be required in a given platform (for example, we don't want the linux implementation of our library to depend on .winmd files)
 - .cs files will start to get bigger and bigger with new platforms and are less readable. 

For these reasons, we opted to instead produce two different implementations, one for Linux and one for Windows. Our repo infrastructure is set such that if a file is named *.Linux.cs it won't be compiled for the Windows implementation and vice-versa for the *.Windows.cs files. We still need to have a PlatformNotSupported version of the drivers that are platform specific, because people will compile against the reference assembly which will have those drivers as existing, so we don't want to throw a "Type not found" exception at runtime when using a windows driver on Linux, and instead we just want to say that this driver is not supported on the platform.

Does that make sense?

shaggygi, 2019-01-07:
Yes, that makes sense.  Thank you.  I originally created this when tinkering around with #110.

closed 2019-01-07 by shaggygi