dotnet/iot #119: Determine structure for SpiController

shaggygi, 2018-12-24:
Similar to the ```GpioController```, there needs to be a mechanism to determine the best ```SpiDevice``` to use.

shaggygi, 2018-12-24:
To be consistent with GPIO and PWM naming structure, should the following files/classes be updated with Driver instead of Device?

```
Spi
  Driver
    UnixSpiDevice.Linux.cs -> UnixSpiDriver.Linux.cs
    UnixSpiDevice.Windows.cs -> UnixSpiDriver.Windows.cs
    Windows10SpiDevice.Linux.cs -> Windows10SpiDriver.Linux.cs
    Windows10SpiDevice.Windows.cs -> Windows10SpiDriver.Windows.cs
  SpiDevice.cs -> SpiDriver.cs
```

Therefore...
```
var connectionSettings = new SpiConnectionSettings(busId, chipSelectLine);
var spiDriver = new UnixSpiDriver(connectionSettings);
byte data = spiDriver.ReadByte();
```

SpiDevice seems like it sticks well, but like GPIO PinNumberingScheme, SpiConnectionSettings is similar to setting up the way the SPI "driver" will work.

IMHO, just seems like Driver would be better at this point to keep consistent with the others (GPIO, PWM, etc.).

joperezr, 2019-01-10:
We can probably bundle this up with #118 since they will be super similar, so I'm fine with closing this one in favor of having only one issue for both Spi and I2c controllers.

shaggygi, 2019-06-15:
@joperezr Please reopen this and assign to me.  New proposal notes are in the works.  Hope to post soon.  Thx.

PR #527 (2019-06-26): Added SPI Create method
Resolves #119

This adds new Create method to return SpiDevice based on OS app is executing.
PR #545 (2019-07-01): Rename SPI Structure
This addresses the SPI updates for moving to Devices folder and updating namespaces..

Resolves #534, resolves #119
closed 2019-06-26 by joperezr