dotnet/iot #124: [Proposal] Provide a new GpioPort class

shaggygi, 2018-12-29:
There are scenarios when a port (with multiple pins) needs to be read from and written to.  It would be nice to have a new `GpioPort` class where you can pass in pin numbers that represent the respective pin/bit for values.

**NOTE**: The code examples are loosely defined only to point out scenarios.
  
## Thoughts/Examples

* `GpioPort` should have a `GpioController` passed to constructor along with pins while instantiating object.  This would work for both dev boards and related device bindings.

```csharp
public class GpioPort(GpioController controller, params int[] bitIndexes)
```
```csharp
GpioController controller = new GpioController();
GpioPort port = new GpioPort(controller, 1, 2, 3, 4);
// or...
int[] bitIndexes = new int[] { 1, 2, 3, 4 };
GpioPort port = new GpioPort(controller, bitIndexes);
```

* Write the value 0x65 to an 8-bit port.
```csharp
GpioPort port = new GpioPort(controller, 1, 2, 3, 4, 5, 6, 7, 8);
port.WriteByte(0x65);

/*
Pin values at this point.
Bit 1; Pin 1: PinValue.High
Bit 2; Pin 2: PinValue.Low
Bit 3; Pin 3: PinValue.High
Bit 4; Pin 4: PinValue.Low
Bit 5; Pin 5: PinValue.Low
Bit 6; Pin 6: PinValue.High
Bit 7; Pin 7: PinValue.High
Bit 8; Pin 8: PinValue.Low
*/
```

* What would be the appropriate methods?  It seems like there should be a variety as there will be different port sizes based on controller or device binding.

```csharp
public bool ReadBit(int bitIndex)
public bool[] ReadBits() // Reads bool values for all assigned pins.
public bool[] ReadBits(params int[] bitIndexes) // Read only pins being requested.
public byte ReadByte()
public ushort ReadUInt16()
public uint ReadUInt32()
public ulong ReadUInt64()
public void WriteBit(int bitIndex, bool value)
public void WriteByte(byte value)
public void WriteUInt16(ushort value)
public void WriteUInt32(uint value)
public void WriteUInt64(ulong value)
public void SetAll(PinValue value);  // Can write all Low or High.
```

* Pins can be in any order, but represent the respective bit starting with least significant bit.
```csharp
// 23 = bit 1; 11 = 8.
GpioPort port = new GpioPort(controller, 23, 2, 9, 26, 1, 19, 10, 11);
```

* Most ports are usually in multiples of 8 (8, 16, 32, etc.).  By allowing the number of pins a user wants to define in a port, you could have cases where they don't equal multiples of 8.  This would be okay as the value could still work with respective pins.  For example, you could have 3 pins that control a [1-of-8 Decoder chip](http://ecee.colorado.edu/~mcclurel/sn74ls138rev5.pdf).  Below shows another example using a 5-bit port.

```csharp
GpioPort port = new GpioPort(controller, 1, 2, 9, 18, 3);
port.WriteByte(0x1A);

/*
Pin values at this point.
Bit 1; Pin  1: PinValue.High
Bit 2; Pin  2: PinValue.Low
Bit 3; Pin  9: PinValue.High
Bit 4; Pin 18: PinValue.Low
Bit 5; Pin  3: PinValue.High
*/
```
* What should happen if value is greater than expected port?  Throw `ArgumentOutOfRangeException` or ignore and only update wired pins?

```csharp
GpioPort port = new GpioPort(controller, 1, 2, 9, 18, 3);
port.WriteByte(0x72);  // 5 bits max value is 0x1F.
```

* Writing values to the entire port will not be synchronous since individual bit indexes have to determine bit/pin, closing/opening pin if needed, and writing pin value to each pin used.  This should be okay as port pins are usually setup and then updated to external hardware using clock/latch lines.

* What happens when user passes same pin more than once?  Just ignore?

```csharp
// Pin 2 used twice.
GpioPort port = new GpioPort(controller, 1, 2, 5, 2);
// port only has 3 pins at this point.
```

* When and how should pin direction be setup?  It seems like it wouldn't be until after the first Read/Write call.
```csharp
GpioPort port = new GpioPort(1, 2, 3, 4);
// No direction/values have been setup at this point.
port.WriteByte(0x06);
// Pins are now outputs with respective value.
```

* Reading or writing pins should always update the direction of pin.  If there are specific pins that need to be inputs or outputs, then those should be left as individual controller pins or separate `GpioPort`.  

    
Other thoughts/concerns?

shaggygi, 2019-02-16:
With the introduction of `IGpioController` and `PinValuePair`, I'm wondering if it would be worth changing proposal slightly to work with them.  Instead of controlling the pins directly, `GpioPort` would be more of a helper between total values & pin values and vice versa.

```csharp
// Example of writing a value to a 3-bit port.

var gpioPort = new GpioPort(1, 5, 8);  //  Create a port using pins 1, 5 and 8 on device.

// This would have overloads with different types like shown above for original Reads/Writes.
// In this case, it could be a byte or larger since it is using 3 bits/pins.
bool result = gpioPort.TryGetPinValues(0x06, out Span<PinValuePair> pinValues);

/*
pinValues at this point.
Bit 1; Pin  1: PinValue.Low
Bit 2; Pin  5: PinValue.High
Bit 3; Pin  8: PinValue.High
*/

// Now you can write those pin values to the controller.
_controller.Write(pinValues); 
```

```csharp
// Example of reading a value from a 5-bit port.

var gpioPort = new GpioPort(1, 2, 9, 18, 3);

_controller.Read(gpioPort.PinValues);

/*
pinValues at this point.
Bit 1; Pin  1: PinValue.Low
Bit 2; Pin  2: PinValue.High
Bit 3; Pin  9: PinValue.Low
Bit 4; Pin 18: PinValue.High
Bit 5; Pin  3: PinValue.High
*/

// In this case, it could be byte or larger since it is using 5 bits/pins.
bool result = gpioPort.TryGetValue(pinValues, out byte value);
// At this point, value = 0x1A;
```

I have not fully grasped the `PinVector` proposal and seems similar, so this change might be overlapping a little.  However, you can have multiple ports on a device (with a mixture of whatever pins and order you wish), even on the same 8-bit ports that most bindings have.

```csharp
var motorPort = new GpioPort(1, 2);
var ledPort = new GpioPort(4, 7, 8);
var decoderPort = new GpioPort(3, 5, 6);
```

You could use for entire port (like on Mcp23xxx) if you wanted, but you'd probably just use the normal Read/Write of register at that point.  Just options :smile:

```csharp
var gpioPortA = new GpioPort(1, 2, 3, 4, 5, 6, 7, 8);  // Or have a helper method to return all for port(s) instead.
bool result = gpioPortA.TryGetPinValues(0x96, out Span<PinValuePair> pinValues);

/*
pinValues at this point.
Bit 1; Pin  1: PinValue.Low
Bit 2; Pin  2: PinValue.High
Bit 3; Pin  3: PinValue.High
Bit 4; Pin  4: PinValue.Low
Bit 5; Pin  5: PinValue.High
Bit 6; Pin  6: PinValue.Low
Bit 7; Pin  7: PinValue.Low
Bit 8; Pin  8: PinValue.High
*/

_mcp23x1x.Write(pinValues);  // Write to all 8-bits on port A.
```

/cc @joperezr @JeremyKuhne thoughts?

shaggygi, 2019-02-16:
Referencing #230 as it appears to be a similar concept.

shaggygi, 2019-02-16:
Another thought... we could actually just get rid of `GpioPort` and add all the logic within PinValuePair using static methods.

```csharp
PinValuePair[] pinValuePairs = PinValuePair.Create(1, 5, 8);  //  Create array using pins 1, 5 and 8 on device.
bool result = PinValuePair.TryGetPinValues(0x06, pinValues);
```

krwq, 2019-02-16:
@shaggygi what about perhaps splitting this into two parts:
- create class which allows looking at GpioController contiguous block of pins as memory (so for i.e.  gpio extenders you don't need to do anything special with indices and they could potentially implement some common interface)
- create a GpioController implementation which allows remapping pins

this way you have 2 classes which allow much more flexibility - the class which provides view could provide similar interface you are proposing

cc: @JeremyKuhne

shaggygi, 2019-02-16:
> * create a GpioController implementation which allows remapping pins
This is somewhat of the idea. The main feature I'm looking for is ability to easily convert a value to (only specific pins) values and vice versa.  Meaning, value 0x0E creates pin values of High, High, Low, High.  With the PinValuePair, the values would be attached to respective pin numbers so you could pass along to any controller implementation's Write method.

The remapping could be accomplished by the order of pin numbers used when using PinValuePair.CreatePairs (or some other better method name :smile:)

I believe additions to PinValuePair can actually solve both (unless there are some other additions you are suggesting I'm overlooking).  I'll update my example above by changing GpioPort to PinValuePair in a few minutes.

krwq, 2019-02-22:
I think we need to update the proposal here to account for PinValuePair

shaggygi, 2019-02-22:
> I think we need to update the proposal here to account for PinValuePair

Discussed https://github.com/dotnet/iot/issues/124#issuecomment-464279785

joperezr, 2019-04-16:
> With the introduction of IGpioController and PinValuePair, I'm wondering if it would be worth changing this proposal to work with them.

Totally agree.

@shaggygi do you know if there is a specific binding that might benefit of this class? I just want to make sure that adding this will be valuable for device bindings.

shaggygi, 2019-04-16:
@joperezr anything that implements IGpioController would benefit including dev boards and bindings like Mcp23xxx, etc. 

Also, I recall @krwq proposing a mapping implementation that works with multiple controllers.

joperezr, 2019-04-16:
Ok, makes sense. Do you mind updating the main porposal to take into account the changes you suggested here https://github.com/dotnet/iot/issues/124#issuecomment-464279785

shaggygi, 2019-04-16:
Sure.  Give me a few days to review and such.

krwq, 2021-01-07:
[Triage] We should fix this as part of GpioPin conversations, closing.

closed 2021-01-07 by krwq