dotnet/iot #126: Should GPIO drivers be renamed to include Gpio

shaggygi, 2018-12-29:
Similar to renaming issue https://github.com/dotnet/iot/issues/123, it seems like the following GPIO drivers would be renamed to include Gpio.  Somewhat redundant since you know in namespace.  However, makes it consistent and could be future drivers (e.g. UnixDriver.Linux.cs in some other namespace later.

| Current | Renamed | 
| ------------- |:-------------:|
| HummingBoardDriver.cs | HummingBoardGpioDriver.cs |
| HummingBoardDriver.Linux.cs | HummingBoardGpioDriver.Linux.cs |
| HummingBoardDriver.Windows.cs | HummingBoardGpioDriver.Windows.cs |
| RaspberryPi3Driver.cs | RaspberryPi3GpioDriver.cs |
| RaspberryPi3Driver.Linux.cs | RaspberryPi3GpioDriver.Linux.cs |
| RaspberryPi3Driver.Windows.cs | RaspberryPi3GpioDriver.Windows.cs |
| UnixDriver.Linux.cs | UnixGpioDriver.Linux.cs |
| UnixDriver.Windows.cs | UnixGpioDriver.Windows.cs |
| UnixDriverDevicePin.Linux.cs | UnixGpioDriverDevicePin.Linux.cs |
| Windows10Driver.Linux.cs | Windows10GpioDriver.Linux.cs |
| Windows10Driver.Windows.cs | Windows10GpioDriver.Windows.cs |
| Windows10DriverPin.Windows.cs | Windows10GpioDriverPin.Windows.cs |

shaggygi, 2019-01-19:
@joshfree @joperezr would it be possible to review this before preview 2?  I think I could complete pretty quickly if there are any decided changes.

shaggygi, 2019-02-01:
@joshfree @joperezr  Just wanted to ping this one again now that Preview 2 is out.

joperezr, 2019-02-12:
Hey @shaggygi thanks for bringing up all of these good questions. I do agree that there are some inconsistencies across our library today, which should get fixed before we ship a stable package. For example, I don't like the fact that Gpio and PWM have a controller which is what you interact with and don't have a specific device object (which in this case would be a GpioPin or PwmPin) but in I2c and Spi we do have the individual devices and don't have a controller for the whole bus. I also think we are lacking some behavior interfaces which should be added which those are tracked by other issues.

In general for your other questions, I don't think that adding Gpio into all of the classnames under Gpio makes sense as since you point out, it does add redundancy when you see the fully qualified name. That said, we do have some counter-examples for that. Let's keep this issue around and use it to discuss concrete examples that should change for the next previews so that we can make things more consistent.

shaggygi, 2019-02-12:
Sounds good.  It might be worth to keep some action items (checkboxes) on this as the main place to confirm the overall API changes that need to happen to call it complete (even if there are other issues created for specific action items).

I agree the namespaces help separate out (although preferred to include).  I just want to make sure (if possible) to make different sections close that make sense.  As noted... I'm okay with System.Device.Gpio.UnixDriver, but we should also have System.Device.Pwm.UnixDriver.

And while using the namespace to separate will help, I'm more worried when we begin working on future interfaces and implement bindings that have multiple functions.  For example, now that the IGpioController is pushed, I plan to begin working on a new binding that provides GPIO and PWM.  The plan is to later add what is pushed for IPwmController (or related solution).  By not having the function in name, it might get a little confusing like the UnixDriver example as we could be using both.

Thanks again for managing this... it's a booger :smile:

shaggygi, 2019-05-27:
@joperezr we might need to get focused on this one soon.  Thoughts?

joperezr, 2019-05-28:
Agreed, all of the issues that are flagged today as 3.0 are things we have to finish in the next few weeks, so we definitely get to this one in the next couple of weeks.

shaggygi, 2019-06-16:
@joperezr I updated the I2C and SPI parts in initial proposal above.  Hope to have more details on #118 and #119 soon.

joperezr, 2019-06-17:
I saw the comment on the I2c proposal about this. I think having that change is fine.

shaggygi, 2019-06-26:
@joperezr @krwq Now that I2C and SPI is winding down, would it be possible to go ahead and put in a PR for updating the related Drivers > Devices folders and namespaces?  I could probably complete those 2 parts of this issue over the weekend.

Related to those, the only thing I had a question on was moving the xxxDevice.cs to the folder.  I like those being grouped with the other ...Device.cs files, but on the fence keeping as is.  And if they are moved in Devices folder, should the namespace be updated for xxxDevice.cs, as well?

krwq, 2019-06-26:
@shaggygi do you mean specifically those 3:

- Folder renamed from "Drivers" to "Devices"
- Namespaces be renamed from "System.Device.I2c.Drivers" to "System.Device.I2c.Devices"
- I2cDevice.cs moved to Devices folder

@joperezr are you ok with that? It looks fine with me

regardless - could you make sure to do any file renames (+ csproj adjustments) in a separate commit to make it easier commit?

shaggygi, 2019-06-26:
@krwq yup, those are the 3 things to focus on for both.  I can do separate PRs for each type if that helps break down since this is breaking changes.

krwq, 2019-06-26:
@shaggygi assuming @joperezr agrees with this work I think ideally I'd at minimum PRs but more depending on the diff size:
- all renames, no namespace changes, don't touch .cs files, only adjust paths in csproj - otherwise the diff might break
- rename namespaces one by one (make sure to update all READMEs and things like that), you can add more to fill up to say 500 lines of code diff

joperezr, 2019-06-26:
@shaggygi those three changes that @krwq mentioned above are approved.

joperezr, 2019-07-09:
I suppose that the PWM PR that we have open is the last thing about this issue that we want to fix? @shaggygi @krwq

shaggygi, 2019-07-10:
@joperezr Still need to move the I2c/Spi Device classes back to their root (based on [PWM discussion](https://github.com/dotnet/iot/pull/576#discussion_r301299089)), but need to finalize #569 before I can update.

I think GPIO still needs a quick review.  For example...
- UnixDriverDevicePin.Linux.cs could have Device removed from name.
- LibgpiodDriverEventHandler.Linux.cs might could move to Drivers folder.
- Still think some files/classes should have a prefiex of Gpio, but I wave the white flag at this point. 😄

joperezr, 2019-07-10:
So for the ones you mentioned, I2c and Spi classes moving won't need to happen before 3.0, because this only involves the specific platform devices, which are internal, so no problem if we don't do that for now.
Same goes for the other cases, UnixDriverDevicePin and LibgpiodDriverEventHandler are internal, so no problem if we want to rename/move them later.

Because of that, I will move this issue to Future for our internal tracking, but feel free to push back if you think otherwise.

krwq, 2021-01-07:
[Triage] I think the main points are already fixed and we do not have enough motivation to continue this effort so closing this issue as fixed.

closed 2021-01-07 by krwq