dotnet/iot #134: [Question] Why are dev board GPIO drivers within System.Device.Gpio API

shaggygi, 2019-01-03:
While it has been extremely easy to get started using GPIO-related features using System.Device.Gpio, I question why include dev board drivers (RPi, HummingBoard, etc.) within the API?  Please forgive my limited knowledge on reasoning, but it seems like this will be an impact in the long term and should be a separation of concern.

For example, I'm assuming when BeagleBone boards become supported, there will need to be BeagleBoneBlackGpioDriver.cs/*.Linux.cs/*.Windows.cs files added within the API.  Is that the correct assumption for other future boards (e.g. RPi4, BeagleBoneGreen, other next popular board, etc.)?

What happens when a dev board becomes unpopular or obsolete?  Do the related files continue to stay within the API for an extended period of time?  This would break someone's implementation if/when they are removed.

What if a company or hobbyist wants to develop their own board with a custom GPIO driver.  For example, there are many companies that reference and design their products based on the BeagleBone open-hardware.  However, their hardware configuration (pins, headers, etc.) are nothing like the layout on the BeagleBone boards.  How would they be able to use some of the APIs that call the `new GpioController();` behind the scenes that pickup "the best driver"?

If the default Rpi3/HummingBoard/BeagleBone GPIO drivers were separated as their own packages, it appears it would still be simple to get started quickly by referencing the GPIO driver package you need and passing it into the GpioController when creating the object.  It would also benefit users by allowing them to reference how the default dev board packages are structured for them to build their own custom GPIO driver packages.  A guidance doc would be helpful in this case, as well.

Again, forgive me on overlooking some of the reasoning and technical challenges.  Just curious on why this direction was decided and what the thinking is when creating new GPIO drivers.

Thanks

shaggygi, 2019-01-26:
This is essentially the same related to PWM, as well.  For example, in the `Windows10PwmDriver` constructor this is references to the HummingBoard.  It seems like this is an unnecessary hard-dependency that should be decoupled.

joperezr, 2019-01-28:
The idea here with GPIO, was to have at least the most common and used boards supported as first-party and that they would ship within the library itself. That was the case for the Raspberry pi, and also the Hummingboard (since that one is heavily used on Azure IoT scenarios). For the rest of the boards, we were thinking on having extension libraries (similar to what we have today with the device bindings package) that would provide support for additional boards, since as you point out, having them on the main library would simply not scale. That said, we are currently investigating on how complete is the support with libgpiod library, since if that has all the features and it performs well, then it wouldn't be very useful to have board-specific drivers, as a libpiod driver would suffice. In that case, then I would agree with removing the board specific ones (RPi and HB) to a different library.

For PWM (I2c and Spi) we only really have two drivers (a Unix generic one and a Windows generic one). I don't expect many more of these to come since they are enough to support a wide range of devices, so I wouldn't really treat those the same as GPIO's case.

joperezr, 2019-02-12:
@shaggygi given that this was just a question, would you be ok if we close it? Or do you think we should try to move the board specific drivers out of the library and use this issue to track that?

shaggygi, 2019-02-12:
Yeah, I think it would be better to rename this and keep if the longer term plan is to remove the board specific code out.

krwq, 2019-02-22:
@shaggygi could you also update the first post so it's easy to read the request after reading 1-2 sentences?

shaggygi, 2019-02-22:
Sure.  My wording can get a little long sometimes.

joshfree, 2019-02-23:
Thanks @shaggygi 😄 

Closing the issue as part of general issue cleanup in the repro

closed 2019-02-23 by joshfree