dotnet/iot #140: Should PinEventTypes be renamed to PinEventType

shaggygi, 2019-01-06:
Seems like the enum [PinEventTypes](https://github.com/dotnet/iot/blob/master/src/System.Device.Gpio/System/Device/Gpio/PinEventTypes.cs) wouldn't be plural.

If so, should others be renamed to be plural?  For example, should [PinMode](https://github.com/dotnet/iot/blob/master/src/System.Device.Gpio/System/Device/Gpio/PinMode.cs) be renamed to PinModes?

buyaa-n, 2019-01-15:
I agree with you PinEventTypes wouldn't be plural

joperezr, 2019-01-15:
@terrajobst what do you think? Do we usually have enums that represent flags as plural in the framework?

joperezr, 2019-01-15:
I checked and most of the framework with similar types do use singular, however, they usually aren't flags and this one is, which is where I'm torn of whether or not this should be plural.

joperezr, 2019-01-15:
> For example, should PinMode be renamed to PinModes?

BTW, PinMode is fine in singular because it is not flags enum, so it will only ever hold one value.

shaggygi, 2019-01-16:
`PinEventTypes` does make sense based on comments.   How about `EventType` in WaitForEventResult.cs?  Should the property be `EventTypes` instead there?

shaggygi, 2019-01-16:
In the [Flags doc](https://docs.microsoft.com/en-us/dotnet/api/system.flagsattribute?view=netframework-4.7.2) example, it is also not plural.  @BillWagner  what are your thoughts?

joshfree, 2019-02-23:
@krwq could you PTAL?

krwq, 2019-02-23:
<details>
<summary>corefx enums with `[Flags]`</summary>

```
System.AttributeTargets
System.Base64FormattingOptions
System.StringSplitOptions
System.Diagnostics.Tracing.EventActivityOptions
System.Diagnostics.Tracing.EventSourceSettings
System.Diagnostics.Tracing.EventManifestOptions
System.Diagnostics.Tracing.EventKeywords
System.Diagnostics.Tracing.EventFieldTags
System.Diagnostics.Tracing.EventTags
System.IO.FileAccess
System.IO.FileOptions
System.IO.FileShare
System.Threading.ThreadState
System.Threading.Tasks.TaskCreationOptions
System.Threading.Tasks.TaskContinuationOptions
System.Threading.Tasks.Sources.ValueTaskSourceOnCompletedFlags
System.Globalization.CompareOptions
System.Globalization.CultureTypes
System.Globalization.DateTimeStyles
System.Globalization.NumberStyles
System.Globalization.TimeSpanStyles
System.Reflection.AssemblyNameFlags
System.Reflection.BindingFlags
System.Reflection.CallingConventions
System.Reflection.EventAttributes
System.Reflection.ExceptionHandlingClauseOptions
System.Reflection.FieldAttributes
System.Reflection.GenericParameterAttributes
System.Reflection.MemberTypes
System.Reflection.MethodAttributes
System.Reflection.ParameterAttributes
System.Reflection.PortableExecutableKinds
System.Reflection.PropertyAttributes
System.Reflection.ResourceAttributes
System.Reflection.ResourceLocation
System.Reflection.TypeAttributes
System.Reflection.Emit.AssemblyBuilderAccess
System.Runtime.Serialization.StreamingContextStates
System.Runtime.InteropServices.DllImportSearchPath
System.Runtime.InteropServices.ComTypes.TYPEFLAGS
System.Runtime.InteropServices.ComTypes.IMPLTYPEFLAGS
System.Runtime.InteropServices.ComTypes.IDLFLAG
System.Runtime.InteropServices.ComTypes.PARAMFLAG
System.Runtime.InteropServices.ComTypes.INVOKEKIND
System.Runtime.InteropServices.ComTypes.FUNCFLAGS
System.Runtime.InteropServices.ComTypes.VARFLAGS
System.Runtime.InteropServices.ComTypes.LIBFLAGS
System.Runtime.CompilerServices.CompilationRelaxations
System.Runtime.CompilerServices.MethodImplOptions
System.IO.FileAttributes
System.GenericUriParserOptions
System.UriComponents
System.Security.Permissions.SecurityPermissionFlag
System.Runtime.Versioning.ComponentGuaranteesOptions
System.Runtime.Versioning.ResourceScope
System.Text.RegularExpressions.RegexOptions
System.ComponentModel.Design.SelectionTypes
System.Collections.Concurrent.EnumerablePartitionerOptions
System.ConsoleModifiers
System.Runtime.InteropServices.TypeLibTypeFlags
System.Runtime.InteropServices.TypeLibFuncFlags
System.Runtime.InteropServices.TypeLibVarFlags
System.Runtime.InteropServices.AssemblyRegistrationFlags
System.Runtime.InteropServices.RegistrationClassContext
System.Runtime.InteropServices.RegistrationConnectionType
System.Runtime.InteropServices.ComTypes.ADVF
[... 79 lines cut ...]
System.Data.CommandBehavior
System.Data.ConnectionState
System.Data.DataRowAction
System.Data.DataRowState
System.Data.DataViewRowState
System.Data.SqlTypes.SqlCompareOptions
System.Data.Common.SupportedJoinOperators
System.IO.MemoryMappedFiles.MemoryMappedFileOptions
System.IO.MemoryMappedFiles.MemoryMappedFileRights
System.Security.AccessControl.AccessControlActions
System.Security.AccessControl.AccessControlSections
System.Security.AccessControl.AceFlags
System.Security.AccessControl.AuditFlags
System.Security.AccessControl.ControlFlags
System.Security.AccessControl.InheritanceFlags
System.Security.AccessControl.ObjectAceFlags
System.Security.AccessControl.PropagationFlags
System.Security.AccessControl.SecurityInfos
```

</details>

Plural: 117, singular: 37

I personally prefer proposed option but for consistency wins so closing...

closed 2019-02-23 by krwq