dotnet/iot #146: Apply comments  from PR #133 related/can be found in SysFs driver implementation

buyaa-n, 2019-01-15:
in LibGpiod driver implementation PR https://github.com/dotnet/iot/pull/133 I had several comments could related/copied from SysFs driver. Apply those comments to SysFs driver and test accordingly such as:

https://github.com/dotnet/iot/pull/133#discussion_r244890760
https://github.com/dotnet/iot/pull/133#discussion_r244891801
https://github.com/dotnet/iot/pull/133#discussion_r245413628
https://github.com/dotnet/iot/pull/133#discussion_r245459095
https://github.com/dotnet/iot/pull/133#discussion_r245459454

joperezr, 2019-04-17:
@buyaa-n is this not done yet?

buyaa-n, 2019-04-18:
Not yet :( ...

buyaa-n, 2019-04-18:
Some of the comments doesn't apply to Sysfs driver anymore and the event handling section for sysfs driver need more refactoring, so am gonna update this issue for refactoring the driver

joshfree, 2019-06-11:
marking this as up-for-grabs in case someone from the community wants to fix this before developers here can get to this issue

PR #606 (2019-07-19): Fixing SetLastError and remove HasFlag to use bitwise instead
Fix #146 

cc: @buyaa-n @krwq 

Addressing comments made on PR #133 on all of our drivers.
closed 2019-07-19 by joperezr