dotnet/iot #156: Main README.md should contain info on how to start with IoT

krwq, 2019-01-23:
This is created based on offline feedback from @Ellerbach.

It would be really beneficial for this repo if main README file contained some more information on how to get a new person on board as fast as possible so that they can try what we have and give us early feedback or contribute.

More specifically:
- how to create a new project, set it up (we already have some bits and pieces of that all over the repo, we just need to gather and better organize them)
- how to publish files and send them to the device
- how to add a new device,
- add some info on what the terminology means and why we chose what we chose
- add link to style conventions

shaggygi, 2019-01-23:
Just a few thoughts.  I'll try to help where needed.  Just let me know. 😄 

1. We now have [How to Deploy an IoT App](https://github.com/dotnet/iot/blob/master/Documentation/How-to-Deploy-an-IoT-App.md) doc that will help in this area.
2. We have the beginnings of a [device binding template](https://github.com/dotnet/iot/tree/master/tools/templates/DeviceBindingTemplate) that will setup a project for a user expected to develop a new binding.
3. I planned to [work on an API doc](https://github.com/dotnet/iot/issues/100) at some point, but this would cover basics down to deeper details on the API.  Waiting for the dust to settle so it doesn't change as much.
4. I think we could improve the samples page, the intro page and roadmap.
5. We really need to decide on a [structure for bindings](https://github.com/dotnet/iot/issues/64) (namespaces, naming conventions, tagging, etc.).  We've done alright so far, but we only have a few bindings currently.  IMHO, I'm not convinced it will be productive/maintainable long-term.
6. We should review and decide on #126.  This will discourage peeps the longer this lingers and breaks more bindings/projects.

Ellerbach, 2019-01-24:
Thanks, makes it much better. Still, there are elements a bit everywhere and it's not that straight forward to start. Starting with a technology and experience you have at the beginning is a critical pass to adoption.
It would be perfect to have a page for very beginners with couple of screen captures from installing VS Code, creating the project, adding the nuget, making sure it's the right .NET Core Framework (so a part of troubleshooting), then taking the blinking led example, building it, and everytime with screen captures.
Something that would target someone who'll just start, has a bit of knowledge but not much. .NET Core is definitely multi platfor, so makes sense to use VS Code for the beginner step by step.
And if there are differencies between platforms (like for example WinSCP to copy on Windows to the device), would be great to have as well few screen captures. 
Some links can point on pages like the one from Scott, that's no issue neither.

And of course, a page, as well with screen captures, for Visual Studio.

For the advance part, most likely no need of screen capture but put few things together would be great. I have to say, I'm usually not ready all open discussions before contributing :-) And I guess, I'm not the only one. But if there is a central page, like the list you've done, with links, even on open issues, I'll read that to continute.

joshfree, 2019-02-20:
@buyaa-n could you help here?

joperezr, 2019-04-17:
I believe this might be a dupe of #100

krwq, 2019-04-17:
yep, looks like a dupe

closed 2019-04-17 by krwq