dotnet/iot #157: Remove DllNotFoundException catching logic after Native Library Loader API deployed

buyaa-n, 2019-01-25:
When loading unix driver we try to load libgpiod driver first, if it is not installed DllNotFoundException catched, in that case we load Sys Fs driver. Remove/update this logic with appropriate one aftet "Native Library Loader API" deployed https://github.com/dotnet/corefx/issues/32015

buyaa-n, 2019-03-14:
"Native Library Loader API" is shipped with 3.0 preview dotnet/corefx#32015
So i have tried to use 3.0 preview for iot project and fix this issue. But it turns out we cannot use 3.0 preview from iot project. The project TargetFramework is set to netstandard2.0 and Santi suggested to not change it until it is released as stable version. I will lookup if I could do some work around but most likely we will wait for full release of 3.0

joperezr, 2019-04-10:
I agree with @safern, In fact even when 3.0 ships, I think we will still want to be able to work in netstandard2.0 so we would either multi-target or stay with netstandard 2.0 target only. What I mean is that I don't believe that this is an important enough reason that would cause us want to change the platform target.

safern, 2019-04-10:
Correct. Actually I meant until .netstandard2.1 is available, rather than .netcoreapp3.0 ships.

krwq, 2021-01-07:
Specific piece of code this is about: https://github.com/dotnet/iot/blob/master/src/System.Device.Gpio/System/Device/Gpio/Drivers/LibGpiodDriver.cs#L81

krwq, 2025-02-06:
[Triage] Closing because it's been open for 3 years and it doesn't seem we care about it enough

closed 2025-02-06 by krwq