dotnet/iot #170: Add bindings for E466616 (NeoPixel Stick - 8x 5050 RGB LED with integrated driver)

krwq, 2019-01-30:


maloo, 2019-02-01:
Hi  @krwq, just found this repo. I was planning to try this out since I'm just using the Neo 8x panel for an embedded system interface. I have to do this within the next week anyway, so if you have any guidelines, APIs or general ideas it would be helpful. I'm planning to use the SPI way of driving them, because I found that to be the only stable (timing) way in linux without writing a kernel driver.
Been using .Net since day one and have a solid background in embedded SW and drivers working for a mobile phone platform company. Let me know if there is anything in the area of drivers/graphics/displays you need help with. An ex coworker is the maintainer of linux gpio so can probably help with that too.
Please let me know how to proceed, first time grabbing ...

joperezr, 2019-02-01:
Hey @maloo, so glad you are joining the community. You can find some docs on how to add a new binding here: https://github.com/dotnet/iot/tree/master/src/devices#contributing-a-binding

I do recommend using [our template](https://github.com/dotnet/iot/blob/master/tools/templates/DeviceBindingTemplate/README.md) when creating the device binding so that you can start directly into coding. I would also recommend to check out [Scott Hanselman's blog post](https://www.hanselman.com/blog/RemoteDebuggingWithVSCodeOnWindowsToARaspberryPiUsingNETCoreOnARM.aspx) for your innerloop testing so that you can test things faster while developing. Other  than that we have a few examples for device bindings in the repo already so feel free to take a look in order to give you a sense of how yours should look like 😄 

Let us know if you have any issues/questions.

krwq, 2019-02-01:
@maloo awesome, thanks for interest in this project :-) other option thank what @joperezr mention is to simply copy the existing project and modify appropriately - please let us know if you find instructions for setup not clear or any other issues you might hit - we are trying to incrementally improve it (so PRs clarifying instructions or issues with a first time experience problems are also welcome).

maloo, 2019-02-01:
First thing I'm wondering about is the classic Copyright/License. Do you have any guidelines for attribution, copyrights etc when porting (as in changing from C++ to C#). I'm used to do stuff from scratch. Couldn't find any existing library ports in the repo at first glance.
Second question is if maybe a library like ADA Gfx should have its own dll/proj.
But first step I probably will just add some code to existing bindings proj.
Also I couldn't find any sln for bindings proj. Are you all using VS Code? I'm used to the VS sln way of working. Like the Gpio proj.

joperezr, 2019-02-01:
For the licensing question, usually BSD and MIT are fine, all other ones depend on case by case and should be reviewed.
Yes, I think the library should have it's own library and dll/proj. We had an issue where we were @shaggygi and I were discussing about where a library like this should live, but for the time being I think a new project under src/devices is fine and then we can move it if we find something more fitting.
We don't have a solution for the bindings because it will be rare for people to want to work with more than 1 binding at a time, which is why instead we only have .csprojs that should work fine with either VSCode or VS. If people find the need to add one device binding solution, we can discuss about adding it in a different issue.

maloo, 2019-02-15:
Are you guys available for comments/review during weekends? Or you respond during MS working hours? Trying to plan my work...
Is it OK to do a mixed PR with multiple bindings/samples and libraries for initial comments? Or do I need to split it before pushing? Easier to work on all dependent parts at once and split in git once done implementing...

krwq, 2019-02-16:
@maloo you might have less comments and depends on the weekend.

We have usually tried adding sample at the sample and README at the same time.

I think at minimum tiny sample is required so that people can know how to use it (you can put the same sample in the README - just add few words what device you are adding and paste the sample)

maloo, 2019-02-17:
Ok, I had some time today, so put together a Neo Pixel binding and sample. Will push it and await feedback before doing more (w1, gfx lib, display bindings).

PR #241 (2019-02-19): Add WS2812B binding
fixes: #170 
cc: @krwq @joperezr
closed 2019-02-19 by krwq