dotnet/iot #172: Add bindings for DS18B20 - Digital temperature sensor

krwq, 2019-01-30:


maloo, 2019-02-01:
Hi, I can help,
I need to write a driver for a MAX31820, I guess these are very similar. Do you have any API. Do you want a 1-wire device and a device driver or just a device driver?

joperezr, 2019-02-01:
Hello @maloo, I know that @ericstj was planning on working on this one as well. He is planning to add support for 1-Wire into our main library. I'm not sure if he is planning to also write a device binding for the actual sensor, so just an FYI so that you don't both end up doing the same work :smile:

krwq, 2019-02-01:
let me update this and other issues so that we don't end up investigating

joshfree, 2019-02-23:
removing the up-for-grabs label as @ericstj grabbed this along with general 1-WIRE support.

@ericstj if your plans change please flip this back to up-for-grabs.  Thanks

MaklaCof, 2019-10-31:
Is there something new. I would like to replace really my old device for measuring hot water in boiler with a new one, based on RaspberryPI. The best suited sensor for this is DS18B20.

maloo, 2019-10-31:
Waiting for @krwq to hit the merge button of #793

krwq, 2019-11-08:
This is merged already, thanks @maloo!

closed 2019-11-08 by krwq