dotnet/iot #174: Add bindings for segment display

krwq, 2019-01-30:
Some examples:
- 1.2" 7-segment LED HT16K33 Backpack
- 1.2" 7-segment display (https://www.adafruit.com/product/1266)
- Dual Alphanumeric Display - Red 0.54" (https://www.adafruit.com/product/1907)

not sure how much they differ, likely code will be very similar, might need some investigation how to write this generic enough

krwq, 2019-02-12:
related: https://github.com/dotnet/iot/issues/179

joshfree, 2019-02-23:
@JeremyKuhne @tarekgh

Ellerbach, 2020-12-22:
@krwq I feel we have all of them now, right?

krwq, 2020-12-22:
We have most popular ones for sure (HT16K33 and TM1637), I'm not sure what driver two bottom ones are using. The last one looks similar to TM1637 but the [datasheet](https://cdn-shop.adafruit.com/datasheets/1266datasheet.pdf) doesn't mention TM1637 (at least not finding it)

krwq, 2020-12-22:
let's close this and create new issue when there is specific request.

closed 2020-12-22 by krwq