dotnet/iot #176: Add bindings for LED Matrices

krwq, 2019-01-30:
This will likely need more digging since there are many different kinds. I.e.:

- 8x8 LED Matrix Square Pixel - Yellow (https://www.adafruit.com/product/1819)
- 8x8 LED Matrix Ultra Bright Yellow-Green (https://www.adafruit.com/product/1045)
- Miniature 8x8 Yellow LED Matrix (https://www.adafruit.com/product/860)
- Mini 8x8 LED Matrix w/I2C Backpack - Red (https://www.adafruit.com/product/870)
- 16x8 LED Matrix Backpack - Green (https://www.adafruit.com/product/2035)
- 32x32 RGB LED Matrix Panel - 4mm Pitch (https://www.adafruit.com/product/607)

related issue: https://github.com/dotnet/iot/issues/11

krwq, 2019-01-30:
@Ellerbach is starting to investigate (can't assign for whatever reason)

mtripsky, 2020-02-26:
@Ellerbach do you need some help? I just joined and looking for some open issues. I noticed that it is one year already since it was created.

Ellerbach, 2020-02-26:
@mtripsky I guess most has been don thru https://github.com/dotnet/iot/tree/master/src/devices#displays.
The 8x8 matrix can be driven by a MAX7219: https://github.com/dotnet/iot/blob/master/src/devices/Max7219/README.md
the large ones 32x32 by the RGBLedMatric driver: https://github.com/dotnet/iot/blob/master/src/devices/RGBLedMatrix/README.md
So my guess is that we should be able to close this one :-)

joperezr, 2020-02-26:
I agree, @krwq feel free to reopen if there is still any display you had in mind that is not yet done.

closed 2020-02-26 by joperezr