dotnet/iot #178: [Question] How will non-binding libraries be managed

shaggygi, 2019-01-30:
While I haven't given it much thought, I wanted to go ahead and create this thread to start some early discussions how we might proceed.  This is based on a particular binding I'm working on, but the issue can be expanded to others as the ecosystem grows.

### Synopsis 
I plan to create a PR in the near future for a SSD1306 OLED display controller.  This binding will initially only include the basics for I2C/SPI communications and commands.  It will also include a method to send data, which is how info is displayed on the screen.  However, it is only for pure data and not for specific text and images.  Meaning, you can't send "Hello .NET" and expect it to magically be the size you want nor the spacing between letters and such.

The way to solve this is to have a separate graphics API that can possibly work with various display controllers in addition to the SSD1306.  I'm referencing the [Adafruit GFX graphics library](https://learn.adafruit.com/adafruit-gfx-graphics-library/overview?gclid=EAIaIQobChMI4uOo4vmV4AIVGYzICh00OwkhEAAYASAAEgJVY_D_BwE) as they appear to have a similar solution.

I have not studied the Adafruit GFX library in detail, but I'm assuming you can create the data bytes needed for text (font, size, attributes, etc.) and images and then pass the data to the specific display controller API.  Ideally, this would be the same concept and we would not need a dependency of the graphics API within the display controller binding API.  (again, not sure yet).

### Question
With having the need of an additional API that could support device binding(s), how will we structure/store these types of APIs?

joperezr, 2019-01-30:
That's a good point. I believe that this additional API should probably live in one of two places: either on the Device bindings package itself (which would mean that we would just have this library live on the repo as if it were a device binding itself) or in a sort of System.Device.Gpio.Extensions library where non-bindings APIs that are used by a lot of bindings are located. 

Another thing we could do, is to jave another subfolder under the devices folder with all of the Common code, where we could place this library so that multiple bindings can just include that code as well as part of their compilation if they need to.

joshfree, 2019-02-23:
@tarekgh @JeremyKuhne

joperezr, 2019-04-16:
For now we have been adding these into Common under devices. That is because at least for now, we only have the one bindings package. Once we move that to one binding-per-package model, we will have to instead have the extensions library where these types of non-binding classes would live in. I'm closing this issue since it was more of a question and I feel that I have answered it with this comment, but @shaggygi feel free to reopen if you still have any questions/concerns around non-binding classes placement.

closed 2019-04-16 by joperezr