dotnet/iot #179: Add bindings for HT16K33

joperezr, 2019-01-30:
This is a 16*8 LED Controller display with keyscan and supports I2c.

joperezr, 2019-01-30:
Here is an interesting link with a similar implementation for Python: https://github.com/adafruit/Adafruit_Python_LED_Backpack/blob/master/Adafruit_LED_Backpack/SevenSegment.py

tawalke, 2019-01-30:
Will work on this one.

krwq, 2019-02-12:
Hey @tawalke, is the branch ready for review yet? What's the remaining work there? :smile:

tawalke, 2019-02-12:
Hi Krzysztof:

The base class lib is completed, I need to publish the changes made to the 7Segment class for this lib so that the branch can be reviewed.  Will do later this week.

gLes, 2019-10-06:
Hi!

As I see this development seems to be stuck in limbo.

I have a working implementation for this controller and with a little cleanup I can put it into a PR in a day or two if you're interested.

krwq, 2019-10-06:
@gLes thanks! Please do, also please check @tawalke's implementation and see if there is anything there you can add to your implementation:
https://github.com/dotnet/iot/tree/015f68cdd5b01a3eeb8ee2043382e0723241f246/src/devices/HT16K33

gLes, 2019-10-06:
Can you give me some preliminary insight on my commit? I tried to follow suite, but it would be nice if someone could confirm I'm on the right path here 🙂

Usage, samples, tests, etc. are coming soon...

krwq, 2019-10-06:
@gLes I haven't reviewed the whole thing, just quickly skimmed and added couple of comments. I think from API perspective you should look at similar device: https://github.com/dotnet/iot/tree/master/src/devices/Tm1637 and see how it was made as an example and do it as similar as possible (if you find that you copy paste a code consider joining some pieces of the code if possible)

krwq, 2021-01-07:
This is already done: https://github.com/dotnet/iot/blob/master/src/devices/Display/README.md

closed 2021-01-07 by krwq