dotnet/iot #186: Library for ADXL345

BrianA-DK, 2019-01-31:
Are there any plans for a library to read a adxl345-sensor via the SPI interface?

I'm trying to write one myself, but I cant quite get the SPI-communication going (I think...)
The following is based on what I could backtrack from samples from the library.
I should read register 0x00 from the adxl345 and return 0xe5, but it only returns 0x00

Sensor and wiring is OK - has been tested by with a C-code example which works.

```
using System;
using System.Device.Spi;
using System.Device.Spi.Drivers;
using System.Threading;
using System.Device.Gpio;

namespace spi_test3
{
    class Program
    {
        static void Main(string[] args)
        {
            var connection = new SpiConnectionSettings(0, 0);
            connection.ClockFrequency = 1000000;
            connection.Mode = SpiMode.Mode3;
	    var spi = new UnixSpiDevice(connection);

            byte[] writeBuffer = new byte[] { 0x00 };
            byte[] readBuffer = new byte[1];
            spi.TransferFullDuplex(writeBuffer, readBuffer);

            Console.WriteLine($"Reg 0x00 = { readBuffer[0] }");

        }
    }
}

```

ZhangGaoxing, 2019-02-01:
Try changing the ClockFrequency? More than or equal to 2MHz

BrianA-DK, 2019-02-03:
Thank you - that got me going 😃 

/Brian

joshfree, 2019-02-20:
Fixed by https://github.com/dotnet/iot/pull/193

closed 2019-02-20 by joshfree