dotnet/iot #194: Nuget links broken

MarcoRossignoli, 2019-02-01:
Nuget source links are broken https://www.nuget.org/packages/System.Device.Gpio
[image]
But package seems ok
[image]

joshfree, 2019-02-19:
@dagood @terrajobst could you give us a quick hand?

dagood, 2019-02-19:
Poking through your official build, it looks like when @joperezr updated the version of Arcade recently, this problem went away. (Extract the nupkg and look at the nuspec.) I know there was an Arcade issue detecting the URL properly in official builds at some point.

Old packages can't be updated (and I don't think there's a specific exception for URLs), but next release will show the right URL.

joperezr, 2019-05-14:
This is fixed now.

closed 2019-05-14 by joperezr