dotnet/iot #195: intel-iot-devkit/mraa and 96boards compatibility/device coverage

IvanFarkas, 2019-02-01:
**[Intel® IoT Developer Kit](https://github.com/intel-iot-devkit) [mraa ](https://github.com/intel-iot-devkit/mraa)and [upm ](https://github.com/intel-iot-devkit/upm)** is a similar library for C++.
Is there any plan to provide compatibility for devices supported by mraa/upm or somehow take advadtage of these libraries?
I'd like to use **.NET Core 3.0 for GPIO, SPI, I2C and PWM** on arm64 Debian on 96boards boards like the **[DragonBoard 410c](https://www.96boards.org/product/dragonboard410c/)** and **[DragonBoard 820c](https://www.96boards.org/product/dragonboard820c/)**.

SergeyRazmyslov, 2019-02-07:
Binding for LCD display from this kit will be added in https://github.com/dotnet/iot/issues/210

joshfree, 2019-02-23:
@IvanFarkas Yes, we have a UnixDriver that uses sysfs and libgpiod under the covers.  Can you try the latest daily build and give us feedback? 

https://github.com/dotnet/iot#how-to-install

/cc @krwq @buyaa-n @JeremyKuhne @tarekgh

buyaa-n, 2019-04-04:
Our generic unix driver should support those boards, but we tested only on Raspberry Pi board.

The generic Unix driver would decide which driver to load at runtime (Sysfs or Libgpiod). Also you can create instance of any those drivers you want to use for GpioController, by passing as constructor argument.
  
Our sysFs driver might not work for some 64 bit Debian version, we have issue created for that https://github.com/dotnet/iot/issues/343, in that is the case you can use libgpoiod driver, but libgpiod driver need install libgpiod library on your pi https://github.com/brgl/libgpiod/blob/master/README

joperezr, 2019-05-08:
@IvanFarkas have you tried this yet? Given that this was more of a question and it has been answered, I'm closing this issue, but feel free to reopen if you think the question hasn't been answered yet.

closed 2019-05-08 by joperezr

ivanfarkas2, 2019-09-11:
Thx. Please close