dotnet/iot #196: Sense HAT

RPagels, 2019-02-01:
Is there support in Dot Net Core C# for the Raspberry Pi Sense HAT?

There is a Python implementation of it here: https://github.com/RPi-Distro/python-sense-hat.  It would be awesome for it to be ported to C#.

Thank you,

Randy

richlander, 2019-02-01:
Not yet. We'd love one. Anyone else want support for this device? Please vote on the issue.

emmellsoft, 2019-02-19:
I have a working C# implementation (currently UWP) that I am porting to .NET Core.
https://github.com/emmellsoft/RPi.SenseHat
Almost completed.

joshfree, 2019-02-19:
Thanks @emmellsoft for looking into porting the SenseHat binding!

/cc @krwq

krwq, 2019-02-19:
@emmellsoft really cool! I've also added some basic implementation in the PR over the weeken: https://github.com/dotnet/iot/pull/242 - I've seen you have a lot of stuff we currently don't have - any chance we could merge this together? (calibration, drawing utilities, more control over the sensors etc)

emmellsoft, 2019-02-20:
Yes of course! Let me just wrap things up a bit!

krwq, 2019-02-20:
Thank you! @emmellsoft let's do it bit by bit - so let's start with my merging my basic stuff, then perhaps calibration and more config options from your library and perhaps changing some conventions

For the drawing utilities we will have a separate conversation on some common interface. @gloveboxes and couple of people here have couple ideas about it and there were couple of proposition for libraries - I don't believe there is a reason to limit to just one library - we can have one for text related stuff, one for lines and simple geometric stuff and others and figure out similarities and right abstraction along the way.

I'd start with the calibration and more configuration because this is lacking completely and also will give you a hang of conventions in our repo - we will need to figure out if we want to put it under devices or tools - depending how you have created your APIs. Let's make sure we provide as easy experience as possible but also have some more powerful slightly more complex options.

emmellsoft, 2019-02-23:
I just pushed my ported code. Although I had totally missed the `System.Devices.Gpio` project, so I based the I2C communication on the `Unosquare` implementation. However, I have limited the usage of I2C, so it should be easy to replace.

I will look into the iot project here -- I would love to contribute! :-)

closed 2019-02-26 by krwq

krwq, 2019-02-26:
Hey @emmellsoft my stuff is merged - feel free to add PRs/issues with anything you find useful and we are missing. I'm really curious to see how far we can stretch sense hat 😄