dotnet/iot #218: We should use project references instead of package references

krwq, 2019-02-11:
Recently we have broken the build but the check in which broke it was green on the CI. This could have been avoided if we have uses project-to-project (P2P) reference instead of package reference. Right now build will be broken only after the package with breaking changes is produced.

Technically we could use the the local nuget package source to be able to use package reference but arcade doesn't support it...

Previous reasoning to use package ref is to allow people to copy paste our projects as an example but I think developer workflow is more important and we can add comments to improve that.

joperezr, 2019-02-11:
Using the local NuGet package would be hard to accomplish. We could instead try using project references instead but it won't be as simple as just changing that, since the main library currently cross-compiles for different RIDs so you would have to specify a RuntimeIdentifier every time you have a P2P. We could solve this by adding some more complex infrastructure like the one corefx has in order to deal with P2Ps, but for now I would keep this issue open and in the backlog, and if we get enough number of hits or it really becomes annoying then we can go ahead and do all the infrastructure work in order to fix this.

krwq, 2019-02-27:
another build break because of this

JeremyKuhne, 2019-02-27:
This is proving to be a time sync, I think we should change it. We may end up being forced to build from source in the future anyway.

joperezr, 2019-04-16:
@krwq pointed out yesterday that now that we've moved to use Arcade for updating our dependencies on the Gpio library, our original desire of having sample projects be copy-pastable is broken so having P2Ps might be considered now. That said, the one good advantage that we still have by using packages is that we still use this as testing and validation that our package is well constructed.

Ellerbach, 2020-12-22:
This has been implemented everywhere in the repo. I'll close this one!

closed 2020-12-22 by Ellerbach