dotnet/iot #226: Update Pca95x4 with IGpioController

shaggygi, 2019-02-13:
@joperezr Go ahead and assign to me.  This will be a good exercise for better understanding on the good work @JeremyKuhne has done.  Most likely, we will have a circle-back to update the I2cController stuff, so this will focus only on the IGpioController for now.

JeremyKuhne, 2019-02-13:
Note that I've got a new library with Pcx857x support in it that I hope to get a PR open for today.

shaggygi, 2019-02-13:
Another side note... we might want to review #135 at some point since expanders support other pin modes in addition to what the current core API includes.  For example, a couple of the MCPs have open-drain outputs.  This might be a good exercise to do sooner than later as I noticed there are some places in the core GPIO that would have to be updated, as well.

JeremyKuhne, 2019-02-13:
https://github.com/dotnet/iot/pull/229 is the initial Pcx857x PR.

Ellerbach, 2021-07-02:
Rather than IGpioController which won't really happen, this should then implement the board abstraction and propoer a real GpioController.

krwq, 2025-02-06:
[Triage] We only should implement GpioDriver

closed 2026-11-05 by pgrawehr