dotnet/iot #230: Add `PinVector` classes for GPIO.

JeremyKuhne, 2019-02-13:
Pin numbers and values can often be stored in bit vectors. This is particularly useful for GPIO expanders where pin numbers are sequential where bit masks can ease storage of state and allow converting directly to data to send to the controllers.

We should add `PinVector32` and `PinVector64` bit structs that encapsulate this concept and allow converting lists of pin/value pairs to bit vectors. (`BitVector32` inspired the name here.)

See example usages in #222 and #229.

``` C#
namespace System.Device.Gpio
{
    /// <summary>
    /// 32-bit vector of pins and values.
    /// </summary>
    public struct PinVector32
    {
        /// <summary>
        /// Bit vector of pin numbers from 0 (bit 0) to 31 (bit 31).
        /// </summary>
        public uint Pins { get; set; }

        /// <summary>
        /// Bit vector of values for each pin number from 0 (bit 0) to 31 (bit 31).
        /// 1 is high, 0 is low.
        /// </summary>
        public uint Values { get; set; }

        /// <summary>
        /// Construct from a vector of pins and values.
        /// </summary>
        /// <param name="pins">Bit vector of pin numbers from 0 (bit 0) to 31 (bit 31).</param>
        /// <param name="values">Bit vector of values for each pin number from 0 (bit 0) to 31 (bit 31).</param>
        public PinVector32(uint pins, uint values);

        /// <summary>
        /// Construct from a span of <see cref="PinValuePair"/>s.
        /// </summary>
        /// <exception cref="ArgumentOutOfRangeException">
        /// Thrown if <paramref name="pinValues"/> contains negative pin numbers or pin numbers higher than 31.
        /// </exception>
        public PinVector32(ReadOnlySpan<PinValuePair> pinValues);

        /// <summary>
        /// Convenience deconstructor. Allows using as a "return tuple".
        /// </summary>
        public void Deconstruct(out uint pins, out uint values);
    }
}
```

JeremyKuhne, 2019-02-13:
@bartonjs, @krwq, @shaggygi, @joperezr 

This is broadly useful when writing device drivers and is also usable in other scenarios. We could make this an internal common thing (with a lot of copies) but I'm pretty confident that it would be good to have public.

While the vectorization of `PinValuePair` isn't super complicated, it is error prone. I'm also expecting that we'll find value in adding manipulation methods to this over time that will also make writing IOT code much simpler.

joperezr, 2019-02-13:
What would Pins uint value mean for each bit? (would a 0 mean that the pin is disabled and a 1 mean that the pin is enabled?)

joperezr, 2019-02-13:
This does work fine in some cases, like the mcp gpio expanders where all pin numers are 0-indexed and the pin numbers are continuous, but that is hardly ever the case for actual GPIO Boards. For this reason, I'm leaning into having these classes live in src/devices/Common and imported by each binding when they see fit. If the main library would use them, then that would be different, but I don't believe that will be the case, which is why I am more inclined to keeping them in src/devices/Common instead.

JeremyKuhne, 2019-02-13:
> What would Pins uint value mean for each bit? (would a 0 mean that the pin is disabled and a 1 mean that the pin is enabled?)

`Pins`: `1` pin has a value, `0` pin doesn't have a value.
`Values` `1` pin is high, `0` pin is low.

> that is hardly ever the case for actual GPIO Boards.

True, but that doesn't preclude them being used in other scenarios. The more important thing here is that this is broadly useful when writing drivers- if I want write my own private driver/adapter/etc. or derive from one of our classes I'm now stuck importing/copying source. To have the bar be that we have to use the API within the main library to include it seems odd to me.

krwq, 2019-02-14:
I agree on the abstraction but I think `Pins` should be `PinMask` since it's more obvious in intention

krwq, 2025-02-06:
[Triage] We didn't have enough need and asks for this in the past 2 years. Closing as won't fix - we can reopen this if neeeded.

closed 2025-02-06 by krwq