dotnet/iot #231: IOT device tests should use MOQ for testing

JeremyKuhne, 2019-02-14:
Using a mocking framework will make tests more robust and easier to follow as frameworks allow easier expressing of intent over making manual mocks.

MOQ is the standard: https://github.com/moq/moq4

Starting Guidelines:

1. _Prefer_ to use MOQ to generate mocks
2. _Prefer_ strict mode mocking (calls to non mocked API throw)

@krwq, @joperezr, @shaggygi

JeremyKuhne, 2019-02-14:
Note: In my first attempt to use this I can't seem to get MOQ to work with `Span<>`. I'm not finding any info on how to set that up. The generated proxy throws an `InvalidProgramException` with `Cannot create boxed ByRef-like values.` when trying to mock `SpiDevice` and call `Write()` on it.

joshfree, 2019-02-19:
@ahsonkhan

Ellerbach, 2021-07-02:
We have quite some examples in the tests where I2C, SPI devices are mocked. That may be enough as, correct, Span<> can't be mocked.

krwq, 2022-04-28:
[Triage] We should talk about this more and decide on the direction

krwq, 2023-09-21:
[Triage] let's close this - it's not actionable for now

closed 2023-09-21 by krwq