dotnet/iot #233: .NET IoT needs a logo

shaggygi, 2019-02-14:
This is more of a fun geek-out than anything, but it seems like this repo/project should have some type of logo to represent.  You can see other project logos on the [.NET Foundation site](https://dotnetfoundation.org/) like Prism, etc.

/cc @joperezr @richlander any thoughts or resources that could design something?

joshfree, 2019-02-19:
@shaggygi any brainstormy ideas for a specific logo that would scale well in size?

shaggygi, 2019-02-19:
Not really, but pinging @jongalloway as he might have some willing/designer resources from the .NET Foundation.

shaggygi, 2019-02-24:
I also tried pinging @BethMassi, but no word back as of yet.

shaggygi, 2019-07-24:
What about the image used in the .NET Core slides?  If so, where can we find it?

[image]

jongalloway, 2019-07-24:
@shaggygi That image looks like it's from shutterstock: https://www.shutterstock.com/image-vector/internet-things-vector-icon-iot-symbol-573776821?src=8x8o92J1ljPmUcgDwHXqGQ-2-65 (see below)

Things that are directly under the .NET brand / marketing at Microsoft are  more restricted on logos, since they fall under Microsoft branding guidelines. @BethMassi?

[image]

BethMassi, 2019-07-24:
The only official logo we are allowed to have is the .NET logo (".NET" in Segoe font in purple box). 
https://github.com/dotnet/swag/tree/master/netlogo 

Are you trying to use an image/icon in a product somewhere to differentiate the workload? We use the Visual Studio design team to do that typically.

shaggygi, 2019-07-25:
> Are you trying to use an image/icon in a product somewhere to differentiate the workload?

Yeah, trying to come up with a good icon for the .NET IoT repo.

Ellerbach, 2020-12-22:
@shaggygi still interested in this? Or should we close this issue?

shaggygi, 2020-12-22:
@Ellerbach I'll go ahead and close.  Never got any traction while trying to reach out to others that might have some designer contacts.  We can open later if needed.  Thx

closed 2020-12-22 by shaggygi