dotnet/iot #234: Add support for 2/3 color e-ink displays

JeremyKuhne, 2019-02-14:
Many of displays are driven by ICs from ILI Technology, such as those from Waveshare and Good Display.

I have a display from Waveshare that is driven by the IL0376F. I'm writing a hopefully generic base driver that can be used to implement the following chips:

IL0326, IL0371, IL0373, IL0376, IL0398, IL3820, IL3829 (and others?)

I haven't done an exhaustive search, if anyone else has links to other chips in this series they would be welcome.

krwq, 2019-02-21:
I've just ordered ILI9341 - not sure if this is the same family - not an e-Ink but the interface might be similar/same - this particular is called TFT LCD display module.

One thing I'm not sure about the stuff from ILI that every page on the datasheet says - actually not sure if I can even copy it - see footer on most of the pages of i.e.:
https://cdn-shop.adafruit.com/datasheets/ILI9341.pdf

this kinda suggest that you can't write code using this spec? technically speaking creating enum with values from such datasheet would be reproducing information there. Not sure if I misunderstand that but would be nice if someone looked at that

krwq, 2022-04-28:
[Triage] It's nice to have issue but considering more than 3 years have passed since this was created and no one requested it closing the issue. We're welcome to take any contributions but we don't need an issue lying around

closed 2022-04-28 by krwq