dotnet/iot #238: Should we add GpioController OpenPin and SetPinMode overloads for multiple pins

shaggygi, 2019-02-16:
The could be for scenarios with bi-directional data, using possible [GpioPort](https://github.com/dotnet/iot/issues/124) or other implementation, where you need to change the `PinMode` when changing direction back/forth on data bus.

```csharp
public void OpenPins(PinMode mode, params int[] pinNumbers)
{
    foreach(int pinNumber in pinNumbers)
    {
        OpenPin(pinNumber, mode);
    }
}

public void SetPinMode(PinMode mode, params int[] pinNumbers)
{
    foreach(int pinNumber in pinNumbers)
    {
        SetPinMode(pinNumber, mode);
    }
}
```

```csharp
// Example of writing data to a 4-bit bus and then swap modes to read.
 var gpioPort = new GpioPort(1, 2, 3, 4);
_controller.OpenPins(PinMode.Output, gpioPort.PinNumbers);  // Open and make all related bits outputs.

gpioPortA.TryGetPinValues(0x06, out Span<PinValuePair> pinValues);
_controller.Write(pinValues);
// Do other logic like clocking data bus, etc.
_controller.SetPinMode(PinMode.Input, gpioPort.PinNumbers);  // Make all related bits inputs.
_controller.Read(gpioPort.PinValues);
gpioPort.TryGetValue(gpioPort.PinValues, out byte value);
```

krwq, 2019-02-16:
@shaggygi GpioPort proposal sounds reasonable and should solve this as well, let me take a closer look at the other issue

shaggygi, 2019-02-16:
Thanks @krwq.  I don’t think this should be limited to just the GpioPort, but on all controllers.  I updated title for IGpioController.  You might have only a few pins on a RPi you want to change quickly, as well.

joperezr, 2019-02-19:
@JeremyKuhne and I discussed this, and we conclude that at least for now we didn't have enough scenarios where you would want to change the pin modes many times and to do it with high performance. Because of this, we decided to for now only allow the Write with  multiple pins because that definitely was a very common scenario and we wanted to be as performant as we could in that case.

@shaggygi do you have a specific scenario where you would want to change pin modes very often?

shaggygi, 2019-02-19:
While I have not had the need on using with bindings yet, I see this being helpful when working with bi-directional data buses to parallel memory, logic latches, etc.  Now that more GPIO controller-types are being added, seems like a "nice-to-have", but not a must.  I guess the individual app could include the code to set the mode in or out when needed.  Maybe keep issue open and add to Future milestone?

krwq, 2019-02-20:
@shaggygi, I'd close the issue for now and reconsider (reopen) the moment someone else creates similar issue or comments here

closed 2019-02-20 by shaggygi