dotnet/iot #249: CharacterLcd.Sample pin numbers do not match with wiring.

SergeyRazmyslov, 2019-02-18:
Hi Guys!

I noticed that the samples for Character Lcd display are not working on Raspberry Pi because the pin numbers do not match with wiring from readme file. I've added fix for the 16x2 display connected via GPIO as part of #248 

Unfortunately I do not have 20x4 display and Mcp23008 extender to check other samples but I think these samples are incorrect as well.

krwq, 2019-02-19:
@SergeyRazmyslov thanks for spotting and sending a fix! Let's fix per case (so let's close the issue when original issue is fixed and file new one when we can confirm it is broken)

monty124, 2019-04-12:
Hi All

I just wanted to make a comment here for anyone else looking, I have a 1602 I2C LCD that uses a PCF8574T piggyback module. To use this with the device binding this is the pinout/working code example:

`
var i2cDevice = new UnixI2cDevice(new I2cConnectionSettings(busId: 1, deviceAddress: 0x27));
var controller = new Mcp23008(i2cDevice);
var lcd = new Lcd1602(registerSelectPin: 0, enablePin: 2, dataPins: new int[] { 4, 5, 6, 7}, backlightPin: 3, readWritePin: 1, controller: controller);
`

took me ages to work this out and a lot of arduino digging around! might help someone else!

apologies if this shouldnt be here, i didnt think it warranted its own issue?

Thanks again everyone!

joperezr, 2019-04-12:
Hey @monty124, thanks for sharing that! Is the PCF8574T the same as an Mcp230008 Gpio expander? If the same binding code works for that other chip, we should probably add that to the README of the Mcp23008 to let people know that they can use the same binding for that type of device.

krwq, 2019-04-12:
@monty124 I concur with @joperezr also since you found it useful perhaps might be worth adding a sample to the LcdCharacter/samples/README.md - do you think you could send a PR with that?

monty124, 2019-04-12:
Hi @joperezr I've taken a quick look at each datasheet and whilst they do look similar and provide what look to be the same functionality, I cannot be certain they are interchangeable or indeed use the same addressing; however; it shouldn’t be too hard to adapt the device binding to work for both in my opinion. I don’t have either IC available bare to wire up and test, to be absolutely sure this would be the way to do it, wire them up; and throw commands at them :)

@krwq sure! I’ve never done this before, let me read up on what’s involved and if you don’t mind, Can I contact you directly for some help? I have another LCD which I suspect may use a different IC; I want to test it on that too and if it is different, I’ll include both. I can also include my example test code in the samples.

[PCF8574_PCF8574A.pdf](https://github.com/dotnet/iot/files/3075470/PCF8574_PCF8574A.pdf)
[MCP23008.pdf](https://github.com/dotnet/iot/files/3075471/MCP23008.pdf)

krwq, 2019-04-12:
@monty124 yes, you can contact me or just send WIP (draft) PR and you can ask questions there :smile:

GTruss, 2019-05-13:
Hi, everyone. Piggy-backing on this thread as I'm not sure it deserves it's own. Has anyone been able to get the LCD on the CrowPi to work? In the CrowPi Python examples, they use:

`lcd = LCD.Adafruit_CharLCDBackpack(address=0x21)
`

After much googling, I thought I had found the .Net Core equivalent:

```
var settings = new I2cConnectionSettings(busId: 1, deviceAddress: 0x21);
var device = new UnixI2cDevice(settings);
            
using (var lcd = new Lcd1602(device)) {
    lcd.Write("Hello World!");
}
```

Unfortunately that doesn't seem to work. I was hoping for a CrowPi example in the Readme doc being as it was mentioned in the text:

> This binding has been tested with a variety of 16x2 and 20x4 displays both in 4bit and 8bit mode and via i2C adapters (such as on the CrowPi). It should work with any character LCD with a 5x8 size character. Common names are 1602LCD and 2004LCD.

I'm pretty new to this so whatever I'm doing wrong is probably something basic. Thanks for your help and let me know if this should go in another thread/forum.

Ellerbach, 2020-12-22:
So as it seems to be solved, closing this issue! Feel free to reopen if needed.

closed 2020-12-22 by Ellerbach