dotnet/iot #258: Iot.Device -> Iot.Thing?

krwq, 2019-02-20:
Should we rename Device to Thing? Since IOT is internet of things and not devices.

cc: @richlander

joshfree, 2019-02-20:
IoT.Thing sounds a bit redundant (Internet of Things Things)

joperezr, 2019-02-20:
Device kind of sounds better to me as well, but that is probably just because I’m used to thinking of them as devices. I don’t have strong feelings against either option.

krwq, 2019-02-20:
I was thinking of 2 namespaces under Iot:
- Thing - devices/sensors
- Net - protocols and way of describing how things to talk to each other

joshfree, 2019-02-20:
I'm not super worried about the name right now given we only have two dozen device bindings under our belt.  I think we'll rename and refactor the namespaces and source code directory structure several times while this project is in the preview phase.

krwq, 2019-02-20:
Closing this issue then as won't fix and we can reiterate later

closed 2019-02-20 by krwq