dotnet/iot #260: Expose public implementation methods via IGpioController

matthewethomas, 2019-02-20:
Please consider adding the following methods to IGpioController. These are all already public methods on `GpioController`'s implementation and will add functionality that is present on the implementation, but missing on the interface.

Pin status/inquiry methods:
---
```C#
PinMode GetPinMode(int pinNumber)
bool IsPinOpen(int pinNumber)
bool IsPinModeSupported(int pinNumber, PinMode mode)
```

Wait methods:
---
```C#
WaitForEventResult WaitForEvent(int pinNumber, PinEventTypes eventTypes, CancellationToken cancellationToken)
ValueTask<WaitForEventResult> WaitForEventAsync(int pinNumber, PinEventTypes eventTypes, CancellationToken token)
```

Callback methods:
---
```C#
void RegisterCallbackForPinValueChangedEvent(int pinNumber, PinEventTypes eventTypes, PinChangeEventHandler callback)
void UnregisterCallbackForPinValueChangedEvent(int pinNumber, PinChangeEventHandler callback)
```

joperezr, 2019-02-20:
This makes sense, although it might be a little complex to implement the eventing methods for some Gpio providers like the MCP devices, @JeremyKuhne any ideas how difficult would that be? I suppose that in some cases where eventing is not supported we might have to do polling on the values to detect changes?

That said, I guess worst case scenario we could have those devices where implementing this is not trivial to simply throw not supported in this case.

shaggygi, 2019-02-20:
You would probably need to poll the interrupt related port bits or could wire up the INTA/B pins.

shaggygi, 2019-02-28:
We probably should add `PinCount` property, as well.

```csharp
/// <summary>
/// The I/O pin count of the device.
/// </summary>
public int PinCount { get; }
```

JeremyKuhne, 2019-02-28:
Related to #295.

These seem like no brainers, I just didn't add them right out of the gate as I didn't need them to solve my immediate problem. 

``` C#
public int PinCount { get; }
PinMode GetPinMode(int pin);
bool IsPinOpen(int pin);
bool IsPinModeSupported(int pin, PinMode mode);
```

Events I'm not so sure about. I'm not sure all devices have the ability to do this in a rational way. If we do add them we'd want some way of tracking whether or not it is possible, is hardware/polling driven, etc.

joperezr, 2019-04-16:
This is approved for everything but Eventing methods. Eventing methods we need to discuss a bit more if we want them part of the interface and simply throw Not Supported on devices that can't deal with this, or to not have them. Flagging as up-for-grabs.

matthewethomas, 2019-04-28:
Awesome!

PR #403 (2019-05-06): Exposed more items in IGpioController
Added more items to IGpioController.  Fixes #260 

Need to review #295, as well.
closed 2019-05-06 by joperezr