dotnet/iot #262: Arduino/Firmata support

techniq, 2019-02-20:
This is more of a question/discussion than an issue, but I was curious if any prior discussions have taken place on supporting Arduinos using [Firmata](https://github.com/firmata/arduino) (similar to how Node does with for example [johnny-five](http://johnny-five.io/platform-support/#arduino-uno)) or [directly](https://github.com/firmata/firmata.js) where you use the Arduino more as a peripheral (with access to it's GPIOs, SPI, etc).  This approach is also used by [gobot](https://gobot.io/) (Go) and [artoo](http://artoo.io/) (Ruby).

Secondly, but tightly related, is the concept of [backpacks](http://johnny-five.io/news/introducing-i2c-component-backpacks/), or custom firmware flashed on an Arduino/etc that you can then communicate with via I2C (or maybe SPI/Serial).  This is an approach when using [node-pixel](https://github.com/ajfisher/node-pixel) to communicate with WS2812 (NeoPixel) lights without an embedded IC.  I saw the [NeoPixel stick](https://www.adafruit.com/product/1426) with integrated IC was [recently supported](https://github.com/dotnet/iot/issues/170), but it would be nice to support non-embedded IC 28xx lights as well (basically all the other lights on [Adafruit](https://www.adafruit.com/category/186) or [Amazon](https://www.amazon.com/s/ref=nb_sb_noss?url=search-alias%3Daps&field-keywords=neopixel).

joshfree, 2019-02-20:
@krwq has been looking into writing a sample of how to flash an Arduino from a C# app running on an Raspberry Pi, but we're not very far along in the thinking in this space.  Happy to have the discussion and get thoughts and ideas here on the topic.

joperezr, 2019-02-20:
If your question is whether or not our main library (System.Device.Gpio) would work on an Arduino, mostly the answer for that is no, and the reason is because Arduino can't run any .NET Standard-compatible framework that we are aware of (.NET Core, Mono, etc). That said, as @joshfree points out, we have been looking into driving an Arduino connected to a Raspberry Pi (which can run our library) for different scenarios.

techniq, 2019-02-20:
@joshfree @joperezr thanks for the replies.  For clarity, the request isn't about running .NET code directly on the Arduino, but interfacing with the Arduino using the [Firmata](https://github.com/firmata/protocol) protocol.  This is the same approach Node, Python, Go, and Ruby have gone (with regards to Arduinos) where you run code (dotnet process) on another machine (i.e. not the Arduino, for example your Windows/Mac laptop, or a Raspberry Pi, but ultimately anything with USB/Serial).  This basically allows you to develop directly on your developer machine (once again, the dotnet process is running on *your* machine) but you are able to leverage the IO on the Arduino (GPIOs, I2C, SPI, etc).

After some quick searches, here are a few .NET libraries (some Framework and some Standard/Core) that use this approach to different levels.

- https://github.com/luisrudge/arduino4net
- https://github.com/SolidSoils/Arduino
- https://github.com/PeterWaher/IoTGateway/tree/master/Things/Waher.Things.Arduino
- https://github.com/lavspent/Lavspent.DaisyChain/

---

Regarding the second part of the discussion regarding "backpacks", this is more of a use case where you would flash custom firmware (could be a custom blend of Firmata) on an Arduino and then interface to the Arduino over I2C (or SPI).  This is useful when you need realtime (i.e. nanosecond) timing such as required to drive a WS2812/NeoPixel, something you couldn't accomplish directly on a Raspberry Pi (well technically it is [supported](https://github.com/jgarff/rpi_ws281x) using some complex direct access.. but the "backpack" is simplier and not directly tied to the Raspberry Pi (ex. flash 1 Arduino, can use on your laptop, Rasberry Pi, Beaglebone, etc).

For a Node example of this, take a look at [node-pixel](https://github.com/ajfisher/node-pixel) and it's [firmwares](https://github.com/ajfisher/node-pixel/tree/master/firmware/build).

---

I'm mostly a hobbiest in IoT, but have created a [Node library](https://github.com/techniq/node-pn532) to communicate with the PN532 NFC chip which is used in a healthcare setting.  I've also research/document a lot of my findings on my wiki.  Here's ones on Firmata and NeoPixel/WS2812 that might be relevant in this discussion.
- https://github.com/techniq/wiki/wiki/firmata
- https://github.com/techniq/wiki/wiki/Neopixels---WS2812

techniq, 2019-02-20:
Just came across this, although this might be native Firmata bindings, although looked relevant at first glance and under the MS umbrella)
- https://github.com/ms-iot/remote-wiring

This was also referenced under @shanselman's [Setting up Windows 10 for IoT on your Raspberry Pi 2](https://www.hanselman.com/blog/SettingUpWindows10ForIoTOnYourRaspberryPi2.aspx) article.

krwq, 2019-02-20:
@techniq - is my understanding correct that what you propose would be perhaps something like IGpioController implementation which could remotely access arduino ports and perhaps allow more than GPIO? That sounds pretty cool - is this something you'd be interested in picking up? 

I couldn't find any license info on https://github.com/firmata/protocol not sure if we can use any of that stuff - we should probably create an issue asking them.

This also doesn't sound like something really hard to implement with some custom protocol in case Firmata doesn't allow this

techniq, 2019-02-20:
@krwq As much as I'd like to help, I'm pretty buried in work (and hobby projects) at the moment.  I'm currently just evaluating dotnet/iot as a replacement for my current (and future) Node IoT projects and saw this as one potential limitation.

Your description is correct, the ability to use dotnet/iot (GpioController, etc) to drive an Arduino "remotely", over a USB to TTL cable (as is typically) such as the [FTDI Serial](https://www.adafruit.com/product/70) (although there are a bunch of different variants of this), but also potentially over BLE (which I've done int he past using Adafruit's [Bluefruit LE UART Friend](https://www.amazon.com/Adafruit-Bluefruit-UART-Friend-Bluetooth/dp/B010M8UZPY/ref=sr_1_3?ie=UTF8&qid=1550700193&sr=8-3&keywords=ble+uart+adafruit).  I think in general Firmata has been traditionally done over Serial (whether USB to Serial, or BLE to Serial) but I think there might be some over I2C now (could be wrong).  I've also seen some interesting serial implementations over MQTT, Wifi, etc (some examples [here](https://github.com/techniq/wiki/wiki/firmata#transports)).  I believe I've flashed an ESP8266 with [Firmata](https://github.com/firmata/arduino/wiki/WiFi-compatible-boards-and-shields#esp8266) and connective via Wifi, but it's been a while (does open up even more possibilities to control something wirelessly).

Currently working on driving a [grid](https://www.amazon.com/gp/product/B01GCPJYD0/ref=oh_aui_search_asin_title?ie=UTF8&psc=1) of WS2812B and have been using node-pixel with the custom firmware/firmata..  Here's a [video](https://video.fric1-2.fna.fbcdn.net/v/t42.1790-2/20999728_671516243057424_7470082351297462272_n.mp4?_nc_cat=111&vs=cf59d7d3775892df&_nc_vs=HBksFQAYJEdEQnVRQUVRMnlDUHZXSUNBQUFBQUFCU0VLdG5idjRHQUFBRhUAABUAGCRHSUNkUHdHaFY3WHZPcGdBQUFBQUFBQV83djlZYnY0R0FBQUYVAgBLAYgScHJvZ3Jlc3NpdmVfcmVjaXBlATAVAAAYCjE2NzAwNjMyNzQWopTh1MqdoiQVAhkFGAJDMxgLdnRzX3ByZXZpZXccF0BB83S8an76GClkYXNoX3YzXzEyODBfY3JmXzIzX2hpZ2hfMy4xX2ZyYWdfMl92aWRlbxEAGBh2aWRlb3MudnRzLmNhbGxiYWNrLnByb2QZHBUAFbyZAgAoElZJREVPX1ZJRVdfUkVRVUVTVBsCiBVvZW1fdGFyZ2V0X2VuY29kZV90YWcGb2VwX2hkE29lbV9yZXF1ZXN0X3RpbWVfbXMNMTU1MDY5OTUwNTQwMAA%3D&efg=eyJ2ZW5jb2RlX3RhZyI6Im9lcF9oZCJ9&_nc_ht=video.fric1-2.fna&oh=43e71dc94f3cee1c79afe998ed58cf19&oe=5C6E0440&_nc_rid=1704ec2a6afbd7f) showing the web interface running on my laptop, which then pushes to a node server and then via node-pixel to the Arduino (and thus the WS2812 grid).

[image]
[image]
[image]

I also have some [strips](https://www.amazon.com/gp/product/B018XAELE4/ref=oh_aui_search_asin_title?ie=UTF8&psc=1) I'm working on using for other effects.

Regarding the license, based on the footnote [here](https://www.arduino.cc/en/reference/firmata), it should be public domain.
> The text of the Arduino reference is licensed under a Creative Commons Attribution-ShareAlike 3.0 License. Code samples in the reference are released into the public domain.

The `StandardFirmata` library/sketch is bundled as part of the Arduino IDE.  The `ConfigurableFirmata` can be created at http://firmatabuilder.com/ and looks to be licensed under [MIT](https://github.com/firmata/firmata-builder/blob/master/LICENSE-MIT)

krwq, 2019-02-20:
@techniq this sounds pretty cool and something we might be interested in. We currently have quite bit of work for other things so for now will mark it as up for grabs. I can definitely help if someone else is interested but might not have to time to do it myself at this point. I'll take a deeper look in a spare moment - if this is easy enough I can just do it at one point.

PR #1039 (2021-02-25): Arduino/Firmata device support
Fixes #262 

This adds support for controlling an Arduino board directly from a PC. The arduino board will run a special firmware called "Firmata", that interprets the commands from the PC and set it to hardware. See Readme for details.

This also includeds an `ArduinoBoard` class, which is a prototype for #878. I'll later push more commits that will show how that would look like with the existing Windows/Linux drivers (but that's somehow also dependent on #1038, because things get complicated otherwise, see #1037 for one of the issues) 

@Ellerbach Can you check [... cut]
closed 2021-02-25 by krwq