dotnet/iot #265: Font 8x8 for Led matrices screens

Ellerbach, 2019-02-21:
Is there somewhere fonts size 8x8 (monochrome) to be used with simple Led 8x8 matrices screen?
Most of the screens, including the existing CharacterLcd are 8x8 or 5x8 or 5x7 size displays for individual characters. Most are monochrome and it would be good to have an official font that can be used with those screens. 
I see multiple options:
* Having a common helper that will load a system found, project it into monochrome bitmaps so it can be served to users who need fonts
* Having a separate class with few fonts with various sizes already hard coded
* A mix of the above
In all cases, fonts need to allow to extend them with own characters. Even the CharacterLcd screens allow to store few characters (5x8 in this case)

krwq, 2019-02-21:
What you probably search for is `texture atlas generator` that can generate bitmap with all characters which can be further used.

Ellerbach, 2019-02-21:
I was more thinking of something that would already exist. So far I'm using 8x8 fonts which are in public domain. It would be good to provide those simple fonts. I'll think about it and see how to generate few quickly based on OS fonts.

techniq, 2019-02-22:
I've used [piskelapp](https://www.piskelapp.com/) in the past to generate pixel artwork as it has a pretty rich editing experience and supports exporting as a byte array, although it [currently](https://github.com/piskelapp/piskel/issues/606) requires a bookmarklet to get monochrome.  Still not exactly what you're looking for, but saw the links @krwq mentioned and thought it was relevant.

You could also load high-res images of the fonts into a `<canvas>` and then downsample it.  This is the [approach](https://photos.google.com/search/led/photo/AF1QipO23gh2Jzz3BuMaeN7CL0ShyJR4tPMsur4WRdn_) I used to downsample pictures to send to a 16x16 LED matrix.  It's been a few years, but [this](https://github.com/techniq/react-pixel-editor/blob/master/src/components/PixelateImage.js) should get you started if you wanted to try that approach.

Ellerbach, 2019-02-22:
I tried to scale down large font size pictures but the result is far to be as good as manual design. I was just wondering if it would be good to have specific 8x8 monochrome font that we can provide. I already started to put together a font like this. Still, it's quite a lot of work.
And thanks for the links!

JeremyKuhne, 2019-02-22:
It would be interesting to see if you can generate anything useful from System.Drawing for this scenario. You'd have to test it out on Windows for now as there currently is a bug in libgdiplus around hinting/antialiasing. https://github.com/mono/libgdiplus/issues/535

I suspect the answer is no as 8x8 is so specific.

joshfree, 2019-02-23:
@safern any thoughts?

Ellerbach, 2019-02-23:
> It would be interesting to see if you can generate anything useful from System.Drawing for this scenario. You'd have to test it out on Windows for now as there currently is a bug in libgdiplus around hinting/antialiasing. mono/libgdiplus#535
> I suspect the answer is no as 8x8 is so specific.

I was already playing wiht the graphic lib to see how to scale down couple of fonts. Result was quite bad but with the option *SingleBitPerPixelGridFit*, the result is much much better! I guess I just need to try to make it on all of the constant size fonts to see if one brings better results. But it means there is a way to automate a bit, at least to export monochrome 8 byte arrays containing all the characters.

Here is the result (zoomed):

[image]

And the code (almost same as from the issue):

```csharp
            Bitmap bmp = new Bitmap(800, 8);
            using (Graphics g = Graphics.FromImage(bmp))
            {
                Font font = new Font("Courier New", 8, FontStyle.Regular, GraphicsUnit.Pixel);
                g.Clear(Color.White);
                g.TextRenderingHint = TextRenderingHint.SingleBitPerPixelGridFit;
                g.DrawString("AaBbCcDdEeFfGgHh", font, Brushes.Black, 0, 0);
            }
            BitmapData bmpData = bmp.LockBits(new Rectangle(0, 0, bmp.Width, bmp.Height), ImageLockMode.ReadOnly, PixelFormat.Format1bppIndexed);

            Bitmap newBitmap = new Bitmap(800, 8, bmpData.Stride, PixelFormat.Format1bppIndexed, bmpData.Scan0);
            newBitmap.Save(@"image.bmp");
```

krwq, 2019-02-23:
I don't think it has to be monochromatic - most 8x8 support shades so that should give perhaps better results

Ellerbach, 2019-02-23:
> I don't think it has to be monochromatic - most 8x8 support shades so that should give perhaps better results. Also I'd try going with size 9/10 and then trim - perhaps that will give somewhat better result

Monochromatic is the mnimum needed. Not all Led 8x8 screens supports PWM. So it has too look great just with 8x8 monochrome. 

Yes, I've been trying 9/10 pixels as well, it actually dépends of the letters. It looks quite hard to generalize and make it fully automated. Take as an example the *G* letter up, it uses already all the 8 width while the *c* or *e* uses only 4. No mentionning that the *g* is already missing its Bottom part.

So all up, I understand why most people design their fonts for 8x8 by hands. I guess in the old Dos version, there must be fonts like that that should be resuable.

techniq, 2019-02-23:
This might be what you're looking for:
https://int10h.org/oldschool-pc-fonts/

tarekgh, 2019-03-07:
There is a variety of BDF fonts out there we can use. It is public domain fonts which is shareable. I already wrote some simple font loader in my work (is not out yet) but I can share what I have so far.

tarekgh, 2019-03-07:
Also, I think we should use BDF as the standard fonts we support instead of defining fonts here and there. of course we can still create out BDF fonts.

tarekgh, 2019-03-07:
https://github.com/Tecate/bitmap-fonts

tarekgh, 2019-03-07:
@Ellerbach please let me know if what I shared is enough in your scenario or you need something more.

Ellerbach, 2019-03-08:
> @Ellerbach please let me know if what I shared is enough in your scenario or you need something more.

Yes! Thanks a lot. It looks like it is fully usable for scenarios for 8x8 font size. Looks like there are enough fonts. And yes, looks like the BDF format can really make sense. So we would just have to add some helpers to manipulate the fonts. Thanks again for sharing!

tarekgh, 2019-03-08:
> So we would just have to add some helpers to manipulate the fonts.

https://gist.github.com/tarekgh/55dfaf0f44689738c3a6ca67941ccdc2 has some initial font loader. it is currently restricted to only fixed char widths fonts which has all characters have exact same size but it is easy to modify to make it support the font in general. I was just trying to get some scenario working for me but later I can polish this more to make it generic. Also, I added some sample how to access the font data there too.

closed 2019-03-08 by tarekgh