dotnet/iot #275: Binding Index needs to be categorized

ZhangGaoxing, 2019-02-23:
Now that **Binding Index** are getting longer and longer, I think it is necessary to classify them in detail. I think of two classification methods.
1. Classification according to the function of sensors
    > #### Environmental Sensors
    >Dhtxx -- Digital-Output Relative Humidity & Temperature Sensor Module
    > ......
    > #### Motor
    > Servo -- Servomotor Controller
    > ......
2. Classification according to the communication mode of sensors 
    > #### GPIO
    > Dhtxx -- Digital-Output Relative Humidity & Temperature Sensor Module
    > ......
    > #### I2C
    > ADS1115 -- Analog-to-Digital Converter
    > ......

Similarly, Binding namespace needs to be categorized too. like
> Iot.Device.Gpio.SensorName
> Iot.Device.I2c.SensorName

> Iot.Device.EnvironmentalSensor.SensorName
> Iot.Device.Motor.SensorName

shaggygi, 2019-02-23:
@ZhangGaoxing somewhat of a duplicate to #63. But I agree, we need to come up with a good solution on this 😄

joperezr, 2019-02-23:
I totally agree with sorting the binding index, but not so much with the namespace idea. The reason is that there are several bindings that support more than one communication protocol, and also have more than one functionality ( for example, sensehat is a temperature and motion sensor, and it is also an led matrix display. Because a binding can only have one namespace, I would keep the namespace as we have it now without a category. For the binding index on the other hand, it should be totally fine to have a binding in more than one category when applicable.

krwq, 2019-02-24:
Closing as dup of #63. I agree this is an issue. Also my PR https://github.com/dotnet/iot/pull/256 tries to introduce more consistency...

closed 2019-02-24 by krwq