dotnet/iot #276: Potential memory leak due to CancellationTokenSource

CosminLazar, 2019-02-23:
While browsing through the source code of GpioController.cs I could see it makes use of CancellationTokenSource without actually disposing it. From what I can see in the source code of CancellationTokenSource it does not have a finalizer (https://github.com/dotnet/corefx/blob/master/src/Common/src/CoreLib/System/Threading/CancellationTokenSource.cs), therefore its Dispose method will never be called.

To make it clear, I have not observed an actual memory leak, I just reviewed the code.
The places that make use of the CancellationTokenSource without disposing it are:
-https://github.com/dotnet/iot/blob/724c8dc7102a1a95e1d70b860b2ad4b3c213387d/src/System.Device.Gpio/System/Device/Gpio/GpioController.cs#L204
-https://github.com/dotnet/iot/blob/724c8dc7102a1a95e1d70b860b2ad4b3c213387d/src/System.Device.Gpio/System/Device/Gpio/GpioController.cs#L234

If time allows I will try to look more into the issue, otherwise I am just letting you know about it.

sharwell, 2019-02-23:
CancellationTokenSource will only leak if it was created with CreateLinkedTokenSource. With that said, it's still a good idea to call Dispose appropriately.

CosminLazar, 2019-02-24:
True, a bad experience with the linked token source must be why I got the alarm in my head when I saw the CancellationTokenSource not being disposed. Furthermore, today I learned that apparently not disposing the CancellationTokenSource can also lead to performance issues https://twitter.com/davidfowl/status/1050278811558850560?lang=en

krwq, 2019-02-24:
@CosminLazar do you feel like sending PR with the fix? We should be disposing it regardless if it will leak or not

CosminLazar, 2019-02-24:
Sure, here it is https://github.com/dotnet/iot/pull/281

closed 2019-03-03 by krwq