dotnet/iot #282: Update compiler

krwq, 2019-02-24:
I've encountered this bug yesterday: https://github.com/dotnet/corefx/issues/35550
when trying to remove some allocations from one of the projects - this is really confusing and much more likely to happen in our repo than in any other (due to writing over SPI/I2C).

We should update our dependencies to make sure that changeset fixing https://github.com/dotnet/roslyn/issues/29092 is included.

cc: @safern

joshfree, 2019-02-25:
Is there a newer drop of buildtools with the updated compiler we can just pick up?

safern, 2019-02-25:
> Is there a newer drop of buildtools with the updated compiler we can just pick up?

I don't think iot uses buildtools at all. 

I guess we will have to do something like we did in corefx to pick up newer compilers:
https://github.com/dotnet/corefx/pull/35134

@krwq please let me know if you need me to send a PR to pick up a new compiler.

closed 2019-03-08 by safern

krwq, 2019-03-08:
Thanks again @safern!

PR #283 (2019-03-08): Update compilers to use the latest 3.0.0-beta3-final
Fixes: #282 

cc: @joshfree