dotnet/iot #287: Add bindings for multimedia abstraction layer (i.e. MMAL, includes Camera/Audio)

krwq, 2019-02-26:
We currently need way of using camera and audio.

One way to provide that is to re-use MMAL effort which allows to access both. It is also used by projects like https://github.com/waveform80/picamera (docs at: https://picamera.readthedocs.io/en/release-1.13/).

We will need to figure out if we want to go even lower with the abstraction or if this is a right spot.

Documentation: http://www.jvcref.com/files/PI/documentation/html/

There is a C# effort already - perhaps makes sense to reuse
https://github.com/techyian/MMALSharp - I've quickly tried it at home but haven't had success (no errors, no expected output)

krwq, 2019-08-08:
cc: @ZhangGaoxing

mtripsky, 2020-02-26:
Is somebody working on this? I would like to help with this

joperezr, 2020-02-26:
I'm not aware of anybody working on this. @ZhangGaoxing added some support for Video on PR #664 but that is just a part of this, so it would be great if you want to contribute with this 👍

joperezr, 2020-02-26:
I'll go ahead and assign you for now, feel free to ask as many questions as you need.

Ellerbach, 2020-12-22:
I feel this is addressed by https://github.com/dotnet/iot/tree/master/src/devices/Media. @krwq maybe we can close this one?

closed 2020-12-22 by krwq