dotnet/iot #288: Ensure SPI/I2C always reads requested number of bytes

krwq, 2019-02-26:
Issue created to track comment https://github.com/dotnet/iot/pull/278#discussion_r259637131 

We should double check if we should not return `int` from the Read which will tell you how many bytes were read.

Usually when reading from Stream the API is allowed to read anything from 0 to requested number of bytes. With I2C/SPI we rely on the fact that we know what the expected number of bytes is.

We need to investigate if that is always the case and if the way we handle it is correct.

cc: @joperezr

krwq, 2019-03-03:
In cases of SPI master controls the clock and when is slave supposed to be writing so the Read should return void since the protocol doesn't give a chance to slave to say that it's finished.

Will look deeper into I2C later.

krwq, 2019-03-23:
just looked at the I2C - the only time you will receive less bytes if slave stretches the clock forever but in terms of I2C this is technically allowed (same for SPI) and is not considered error (although would be rather unexpected). Since master controls the clock and slave is supposed write as long as SDA is up that means that slave must write as many bytes as user expected and if it doesn't then you will get bogus data but there is no way you can know about it (at least on the first abstraction layer).

closed 2019-03-23 by krwq

raffaeler, 2019-05-09:
Question: what if the device can send a variable number of bytes?
The Linux I2C driver returns the number of read bytes for this reason and IMO it makes sense because it's an application protocol decision, not a hw peripheral specification.

joperezr, 2019-05-09:
@raffaeler can you provide an example of a case where the number of bytes is variable? AFAIK all of the device bindings we've seen and all the libraries similar to GPIO will know how many bytes to read for every I2c call. There are some devices who might send a variable number of bytes of data, but usually for these you first would make an I2C call which will return you 1 byte telling you the size, and then you can make your second I2C call querying for that number of bytes of data.

raffaeler, 2019-05-09:
Hi @joperezr, not yet.
I am translating the Flick C library using wiringPi and the Linux i2c library, but the source code does not mention how many bytes are expected nor there is any comment about the length of the packets.
The C library just read data in the buffer and use the length as a discriminator.

I guess the current C# I2C library will throw for a timeout if the buffer is not entirely filled. Am I right?

I am just afraid that any device relying on an extrernal line (such as the one I am using) or simply a timeout, would be more difficult to use.
Changing the API now is simple, but should any device work as I am probably seeing, could introduce an annoying breaking change.