dotnet/iot #295: Clarify GPIO interface / members

JeremyKuhne, 2019-02-28:
After discussing with @bartonjs, this is likely a better starting point for this API.

``` C#
public interface IGpioController
{
        void OpenPin(int pin, PinMode mode);
        void ClosePin(int pin);
        void SetPinMode(int pin, PinMode mode);
        void Write(int pin, PinValue value);
        void Write(ReadOnlySpan<PinValuePair> values);
        PinValue Read(int pin);
        void Read(ReadOnlySpan<int> pins, Span<PinValuePair> values);
}
```

There are couple of things here.

1. We don't want to use identifiers that are longer than they need to be to convey full context/meaning (don't want to be unnecessarily wordy).
2. Splitting out the multi-pin Read() for clarity - this method should throw if `pins` / `values` are not the same length.

This would also be followed with changing `pinNumber` to `pin` in the rest of the `GpioController` class public interface.

cc: @krwq, @joperezr, @shaggygi 

(@shaggygi, sorry, was in the midst of reviewing this in detail when #239 got merged 😭)

shaggygi, 2019-02-28:
@JeremyKuhne is this something that will be added soon?  I just started updating #226, but can hold off until then.

JeremyKuhne, 2019-02-28:
> @JeremyKuhne is this something that will be added soon? I just started updating #226, but can hold off until then.

You don't need to hold off.

krwq, 2019-03-01:
@JeremyKuhne why `void Read(ReadOnlySpan<int> pins, Span<PinValuePair> values)`; and not:
`void Read(ReadOnlySpan<int> pins, Span<PinValue> values);`? Same when writing why not split numbers and values? This can allow you to create multiple patterns of values (i.e. for stepper motor you can save the whole sequence without double storing pin numbers)

shaggygi, 2019-03-01:
I haven't thought about this much, but why not return like...

`ReadOnlySpan<PinValue> Read(ReadOnlySpan<int> pins)`

bartonjs, 2019-03-01:
@shaggygi That provides an unclear ownership/lifetime/validity semantic.  Read-into-this-Span is "I own this memory, please write to it".  Read-and-here-is-a-Span says "the answer was written somewhere, here's a way to access the value, please copy this out before you call Read again".  So

```C#
Span<PinValue> values = new PinValue[pins.Length];
Read(pins, values);
Read(pins, values);
```

Clearly has replaced the first answer with the second.

```C#
ReadOnlySpan<PinValue> firstRead = Read(pins);
ReadOnlySpan<PinValue> secondRead = Read(pins);
```

It's not obvious that `firstRead` and `secondRead` will have the same values now, which may have been different than before the second call to `Read`.  (And if they don't have the same values (because Read is allocating new memory for the return), it should just return the array instead of a span over the array).

shaggygi, 2019-05-16:
This could probably be closed now.

joperezr, 2019-05-16:
I agree. @JeremyKuhne feel free to reopen in case you think we are still missing something from the interface.

closed 2019-05-16 by joperezr