dotnet/iot #298: Support for Imperial vs Metric systems etc.

CrustyJew, 2019-03-01:
as @krwq mentioned [here](https://github.com/dotnet/iot/pull/256#discussion_r258229670), it would be nice to iron out some locations for and how we'd want to support multiple measurement systems.

I'm looking at the HCSR04 device binding right now, and something like a public enum that can be reused to specify which kind of measurement you want back would be cool. That way you can pass it in on initialization and have it calculated in that format so you don't get any weird issues or lose precision while doing conversions.

Would this enum go in a new file along side `PinMode` maybe?

krwq, 2019-03-01:
2 issues here:
- internal units of the device do not match exposed API and we want to avoid double conversion
- API for conversion

I think we should create a separate one for each of those problems or drop one of them in here.

Let's start with first:
@CrustyJew for many of the sensors `double` gives enough precision that you won't lose much by double converting or neither that is a large overhead (few processor cycles) - also most of the sensors already internally use the units we have picked to use in our repo (Celsius, %rH etc). I think once we get the first device which internally stores the value in say inches instead of cm/m then I'd initially add the enum over there and then when another device shows up then possibly leverage that API further and move it to i.e. Common directory - after that we should consider making it public.

Adding say `Unit` enum to every constructor to every device sounds like it may pollute a lot of classes with a lot of logic which in most of the cases won't be used - I'd prefer to have it only when there is a good reason to have it (i.e. very high precision thermometer which internally stores Fahrenheit - not sure if such exist - just saying as the example) - but then perhaps might be worth considering using different type than double for such cases.

As for the conversion issue:
I personally think the interfaces are the way to go and address this with static extension methods. I initially got some push back on that but will try to get some agreement or perhaps some alternative solution for that.

Perhaps there is some intermediate solution for both issues and design number type which includes a unit and possibly allows representing values as i.e. fractions or other alternative formats - this way we could avoid any conversions everywhere and do it only when data is requested in specific format..

CrustyJew, 2019-03-04:
True, most internal memory on these devices probably is in metric. I was thinking more in terms of the sonar sensor where all it is is an amount of time that a pulse took to reach a destination and return. So you have to convert the time into distance based on your own math.

I could see the desire/need to be able to actually pass in a temperature measurement to make the sonar more accurate based on a more accurate speed of sound calculation.

Would it maybe make more sense to have more generic objects like `Temperature`, `Distance`, `Velocity`, etc and have some way to initialize them with whatever unit the device measured it with, and have properties to convert or store the conversions to other units as needed?

krwq, 2019-03-04:
@CrustyJew from modelling perspective that would be ideal but from usability and perf point of view it may not be ideal...

krwq, 2019-03-05:
@CrustyJew we ended up doing that 😄

krwq, 2019-03-05:
So this issue is partially addressed (only temperature for now)

CrustyJew, 2019-03-06:
Ha nice! Yeah something like that should work pretty well I think. I'll see if I can poke any holes in it here in a bit.

krwq, 2019-03-06:
@CrustyJew thanks! We will wait a bit before adding more units (next one will likely be length units since they also have multiple units). Once we decide that this is what we want long term we will figure out if we want to have all units or just the ones which have multiple units. (i.e. for humidity seems like everyone uses %rH so does not feel like it is strictly needed - might be good to have for parity but we are not sure if this is what we want yet)

krwq, 2019-03-22:
So we currently got only 2 sensors
- HCSR04
- VL53L0X (https://github.com/dotnet/iot/pull/312)

so far the units don't match (micrometers vs cm) but 2 sensors is probably too little to get into any conclusions. Seems we will need to consider having also multipliers for some of the units at one point (Mili, One, Kilo, Mega etc)

CrustyJew, 2019-03-22:
At some point in the near future I'm looking to use a HX711 and I'll have to get some code in there for that. That'll be a real good candidate for this as it basically needs to take a raw input signal and calibrate, and convert it to a weight measurement based on that calibration (could be any measure of weight).

CrustyJew, 2019-04-09:
@krwq I haven't finished all the documentation for the HX711, but I did an initial implementation of it and included the idea I had for another `Unit` type for weights. 

The commit has all the HX711 code as well, but I changed the `Units` project as well with the weight implementation.

@ZhangGaoxing I'd like to hear your thoughts as well. I added specific methods to only get one weight measurement system to the hx711 sensor as well to avoid doing extra math if you don't need the conversions.

https://github.com/CrustyJew/iot/commit/a18dc63d693b355b394980480f09137dd6aa405c

Ellerbach, 2020-12-22:
With support of UnitsNet everywhere, we do support any kind of unit. So closing this issue.

closed 2020-12-22 by Ellerbach