dotnet/iot #301: Using Static Extension Method to Provide Data Conversion

ZhangGaoxing, 2019-03-02:
To fix this https://github.com/dotnet/iot/pull/291#issuecomment-467884576 , I think we should provide different extension methods according to different classes. Take temperature sensor as an example:
1. First, temperature sensors should inherit a unified interface `ITemperature`. @krwq mentioned it #256 
```C#
public interface ITemperature
{
    double Temperature { get; }
}

public class Lm75 : ITemperature, IDisposable
{
    // in celsius
    public double Temperature { get => GetTemperature(); }
}
```

2. Second, provide a temperature static extension class. Namespace should be `Iot.Device.Extension.TemperatureExtension`
```C#
namespace Iot.Device.Extension.TemperatureExtension
{
    public static class TemperatureExtension
    {
        public static double CelsiusToFahrenheit(this ITemperature temperature, double celsius)
        {
            return celsius * 1.8 + 32;
        }

        public static double CelsiusToKelvin(this ITemperature temperature , double celsius)
        {
            return 273.15 + celsius;
        }
    }
}
```

3. Finally, it should be used like this
```C#
using Iot.Device.Extension.TemperatureExtension;

static void Main(string[] args)
{
    I2cConnectionSettings settings = new I2cConnectionSettings(1, Lm75.DefaultI2cAddress);
    UnixI2cDevice device = new UnixI2cDevice(settings);

    Lm75 lm75 = new Lm75(device);

    double fahrenheit = lm75.CelsiusToFahrenheit(lm75.Temperature);
    double kelvin = lm75.CelsiusToKelvin(lm75.Temperature);
}
```

---

What do you think? @krwq @shaggygi @joperezr @JeremyKuhne 
(Who should I @?😄)

shaggygi, 2019-03-02:
Understanding this is just a brief overview concept, we could add other conversions Fahrenheit > Celsius, Kelvin > Fahrenheit, etc.

I like how it is an extension, but question is it really needed where we could use the static methods in a common helper like...

```csharp
// NOTE: Namespace could be better; I just used as example.
double fahrenheit = IoT.Device.Common.Temperature.CelsiusToFahrenheit(lm75.Temperature);
```

krwq, 2019-03-02:
I'd go with just `ToKelvin`, `ToFahrenheit` and assume Celsius

krwq, 2019-03-05:
I think I'll close this issue since we went with the structures. note that this does not exclude possibility of using ITemperature interface in the future.

Please let me know if there is anything more we would like here and re-open if so.

closed 2019-03-05 by krwq