dotnet/iot #315: Reconsider namespaces

krwq, 2019-03-05:
I've heard couple of request to change some of the namespaces over several days.
We should reconsider doing some changes in this area.

Here are some examples I heard:
- move I2c/Spi and other protocols outside of Gpio and put them under `*.Protocols`
- change `Iot.*` to something under `System.*`
- consider changing `*.Devices.*` to something else (i.e. SoftPwm does not fit there but there is not better place to put it) - one proposition I've created some time ago was `Things`

joshfree, 2019-03-05:
@krwq could you link/quote the relevant feedback?

krwq, 2019-03-05:
@joshfree this was various hallway/IM conversations - I referenced the recent PR feedback which mentions this above.

joperezr, 2019-04-16:
I'm not sure about this one either. @richlander what do you think? Are you happy with the current state of our namespaces?

terrajobst, 2019-04-16:
Was it anyone on @dotnet/fxdc? Namespace choices are hard and making everyone happy with be hard. I'm happy to help here.

joperezr, 2019-04-16:
Currently we have the same namespaces as the ones that were discussed and approved in the API review. @krwq are there any concrete changes that were proposed here? So we can review them with @terrajobst.

krwq, 2019-04-16:
See bullet points in the issue description:
- create namespace for protocols i.e. (Iot.Protocols/System.*.Protocols)
- move Iot namespace under System. I.e. System.Iot
- move stuff which is not a device outside of System.Device

terrajobst, 2019-04-16:
If you (the IOT team) believes, that's the right thing, make a proposal and we can review it. I'm also happy to meet and chat if you want an opinion on namespace factoring, based on the concerns that were brought up.

shaggygi, 2019-04-17:
@joperezr @terrajobst over the weeks, I read various mentions of API reviews for this repo.  Is the intent to collect a list of issues/PRs to discuss/design/code or are there meetings that have already occurred?  It would be nice to have those streamed or posted so the community can understand some of the opinions and decisions.  The 3 videos from initial concept were insightful.

joperezr, 2019-04-17:
@terrajobst I'll chat offline with @krwq so that we can come up with an agreement since I don't believe we should change them, but we will get a consensus and then we'll draft a proposal if needed.

@shaggygi we haven't really had any Api Review discussions yet for the iot repo other than the 3 initial ones. The reason for that is that since we are still in prerelease, we are still allowed to add API without needing to follow the full process as long as the API to be added is simple and we all agree that it is needed. For more "controversial" proposals, or once we ship 1.0 we have to follow the full review process. The idea of the full process is that once we have some proposals ready for review by the review panel, we will notify @terrajobst and he will get us on the list of one of his public reviews. These ones are streamed via Youtube, and contributors may participate as well. Is there a particular api proposal in the repo that you care about? That way I can re-check it and see if it's ready for review.

shaggygi, 2019-04-17:
@joperezr sounds good.  Being an outsider, some of the wording sounded as there were other meetings going on.  So my bad of interpretation.  The only issue I can think of is possibly #126.

joperezr, 2019-04-17:
Sounds good, sorry for not being clear about the process. I'll take a look at that issue again and submit it to the review process if its ready.

joperezr, 2019-05-14:
Clos

closed 2019-05-14 by joperezr