dotnet/iot #319: [Question] System.Device.Gpio.dll not loaded on classlib

DanielSSilva, 2019-03-08:
For some reason ( I might be missing something), if I have a class library that has the System.Device.Gpio package as requirement, when I publish it for the linux-arm runtime the dll is not generated on the publish folder. If I add the OutputType to exe and change targetFramework to netcoreapp2.0, the dll is generated on the publish folder. Is this the expected behaviour?

krwq, 2019-03-09:
Is this specific to only this package? If not then I'd recommend moving this issue to https://github.com/dotnet/cli

joperezr, 2019-03-11:
A classlib targetting netstandard2.0 can’t really be published for a runtime since it won’t actually be runnable. If you take a look at the depa.json file that gets generated when you try this it will call out that you depend on system.device.gpio correctly. 

The way this works is that once you build a runnable app (an exe targetting netcoreapp) that depens in your classlib, both the classlib and system.device.gpio will succesfully be copied to the published folder since that is a runnable framework.

Does that make sense?

DanielSSilva, 2019-03-12:
Thank you for the clarification @joperezr . It does make sense. The reason why I was asking this is because I wanted to create a classlib that initializes a bunch of sensors (raspberry sensehat) and wanted to use that with powershell, by loading the assembly, but it thew an exception saying that gpio dll was not found. Changing it to an exe generated the dll, as you've explained.
Thank you again for your clarification

closed 2019-03-12 by DanielSSilva