dotnet/iot #320: Support PN532

techniq, 2019-03-09:
Would be nice to add support for the PN532 (RFID) module.  I wrote a pure (non libnfc) Node.js [implementation](https://github.com/techniq/node-pn532) a few years ago that could be a good reference, although I mainly worked with Serial and only partialy support I2C (and no SPI) (which it supports all 3).  Seems like starting with I2C would be the recommend bus if we only handle 1 initially.  [Adafruit](https://github.com/adafruit/Adafruit-PN532) has a good C++ library, as does [SeeedStudio](https://github.com/Seeed-Studio/PN532).

I might try to tackle this myself sometime, but I at least wanted to document an issue incase someone handled it quicker.  I also saw the [MFRC522](https://github.com/dotnet/iot/pull/293) PR and didn't know if some reuse could be shared between these.

For parsing NDEF messages I used [ndef-js](https://github.com/don/ndef-js).  There is a [C++](https://github.com/don/NDEF) library by the same developer, and I just came across [this](https://github.com/andijakl/ndef-nfc) one that include C# that might come in handle when dealing with card/tag data.

- [Datasheet](https://www.nxp.com/docs/en/nxp/data-sheets/PN532_C1.pdf)
- [User guide](https://www.nxp.com/docs/en/user-guide/141520.pdf)

krwq, 2019-03-09:
If there is possibility of sharing implementation with MFRC522 that should be ideal - if not then at least we should have the same public APIs.

Ellerbach, 2019-07-02:
I've started some work on the PN532. Hzere isthe current work: https://github.com/Ellerbach/Pn532. Working good to read and write Mifare cards. Implementation is low level one so far, so mainly the basic registers to be able to detect, read and write cards. Plus authentication.
Implementation is still very early. Feedbacks and help welcome.

Ellerbach, 2020-12-22:
@techniq full support for PN532 now. Check it https://github.com/dotnet/iot/tree/master/src/devices/Pn532

closed 2020-12-22 by Ellerbach

techniq, 2020-12-22:
@Ellerbach Thanks!